openwrt/packages #8906: gawk: update request to latest 5.0.0 version

dnmTX, 2019-05-05:
Description:
Thank you all for maintaining that excellent project.
I'm using Entware on my DD-WRT router and gawk helps a lot with some of my custom scripts there. New version is up and judging by the release notes i think it's worth updating:

> Changes from 4.2.1 to 5.0.0
---------------------------
1. Support for the POSIX standard %a and %A printf formats has been added.
2. The test infrastructure has been greatly improved, simplifying the
   contents of test/Makefile.am and making it possible to generate
   pc/Makefile.tst from test/Makefile.in.
3. The regex routines have been replaced with those from GNULIB, allowing
   me to stop carrying forward decades of changes against the original
   ones from GLIBC.
4. Infrastructure upgrades: Bison 3.3, Automake 1.16.1, Gettext 0.19.8.1,
   makeinfo 6.5.
5. The undocumented configure option and code that enabled the use of
   non-English "letters" in identifiers is now gone.
6. The `--with-whiny-user-strftime' configuration option is now gone.
7. The code now makes some stronger assumptions about a C99 environment.
8. PROCINFO["platform"] yields a string indicating the platform for
   which gawk was compiled.
9. Writing to elements of SYMTAB that are not variable names now
   causes a fatal error. THIS CHANGES BEHAVIOR.
10. Comment handling in the pretty-printer has been reworked almost completely
    from scratch. As a result, comments in many corner cases that were
    previously lost are now included in the formatted output.
11. Namespaces have been implemented! See the manual.  One consequence of this
    is that files included with -i, read with -f, and command line program
    segments must all be self-contained syntactic units. E.g., you can no
    longer do something like this:
           gawk -e 'BEGIN {' -e 'print "hello" }'
12. Gawk now uses the locale settings for ignoring case in single byte
    locales, instead of hardwiring in Latin-1.
13. A number of bugs, some of them quite significant, have been fixed.
    See the ChangeLog for details. 

```
Thank you for considering 👍

diizzyy, 2019-05-12:
Done as of https://github.com/openwrt/packages/commit/8199a4b8ab3105d51c8b6267b59947bf5d3b8d41

closed 2019-05-12 by diizzyy

dnmTX, 2019-05-12:
@diizzyy  thanks a lot 👍