openwrt/packages #8915: strongSwan: How can I set the timer of retransmit of initiator request?

qiuhez, 2019-05-06:
I'm testing to start an ike negotiation. 

The log says,
"May  6 02:41:25 bgw7vm011 strongswan: 12[NET] <scm-bgc-0|1> sending packet: from 172.38.200.1[500] to 172.38.1.131[500] (464 bytes)
May  6 02:41:25 bgw7vm011 strongswan: 04[NET] error writing to socket: Invalid argument"

and the next time to send this packet is after 4 seconds:
"May  6 02:41:29 bgw7vm011 strongswan: 06[NET] <scm-bgc-0|1> sending packet: from 172.38.200.1[500] to 172.38.1.131[500] (464 bytes)"

Can I configure the timer to a shorter interval? How? 
Thank you.

lucize, 2019-05-07:
I think the best place is https://wiki.strongswan.org/projects/strongswan/issues or a mailinglist

stintel, 2020-01-22:
> I think the best place is https://wiki.strongswan.org/projects/strongswan/issues or a mailinglist

Or IRC or the OpenWrt forum.

Closing, not a bug, the issue tracker is not for support questions.

closed 2020-01-22 by stintel