openwrt/packages #8926: snmpd crash loop

lleachii, 2019-05-07:
Upgraded to OpenWrt SNAPSHOT r9945-bc85640cdc, I'm now receiving:

`Tue May 7 15:04:40 2019 daemon.info procd: Instance snmpd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash`

Snapshots a few days ago worked.

diizzyy, 2019-05-07:
Platform? Any changes to default config?

lleachii, 2019-05-07:
- Installed on apm821xx.
- I'm using default snmpd file from opkg.

lleachii, 2019-05-08:
- I was unable to reproduce the issue on another apm821xx device
- Re flashing the same device with r9945-bc85640cdc and reinstalling seems to have solved the issue

Thanks for your time and attention.

lleachii, 2019-05-28:
- This has reoccurred using r10084-b97d44af67
- This time, I **am able** to reproduce the issue on an identical device
- Re flashing does not solve the issue this time

```
root@OpenWrt:~# opkg list-installed | grep snmp
libnetsnmp - 5.8-1
snmpd - 5.8-1
```

lleachii, 2019-05-29:
https://forum.openwrt.org/t/snmpd-crash-loop/37779

lleachii, 2019-06-03:
Also occurs on OpenWrt SNAPSHOT r10124-405e08bee6

lleachii, 2019-06-04:
No crash loop with OpenWrt SNAPSHOT r10127-3209f5ae3d

diizzyy, 2019-06-22:
Closing this as it's resolved

closed 2019-06-22 by diizzyy

lleachii, 2019-07-17:
Upgraded from OpenWrt SNAPSHOT r10325-b16faf9bc7 (working)

Now not working: OpenWrt SNAPSHOT r10494-e05310b9b8

diizzyy, 2019-08-02:
Hmm... I wonder if it's some kind of race condition as it seems to occur randomly. Could you try compiling r10325 and r10494 by hand and see if it you get the same results as it may be a specific buildbot that's causing the issues.