openwrt/packages #8927: ddns-scripts: configurable HMAC algorithm in update_nsupdate.sh

laurikimmel, 2019-05-07:
Maintainer: @chris5560
Environment: OpenWrt 18.06.2, ddns-scripts - 2.7.8-1, ddns-scripts_nsupdate - 2.7.8-1

Description: there is no configuration option to set HMAC algorithm for ```nsupdate``` request. By default  ```nsupdate``` is using ```hmac-md5``` which is not working in case ddns server is expecting something else. For example ```hmac-sha256```.

dibdot, 2019-05-13:
According to the sourcecode, ddns-scripts is using the following nsupdate syntax: `key $username $password`. Does it work if you set the 'username' option to 'hmac-sha256: <keyname>' in /etc/config/ddns? (as explained here: https://bind.isc.org/doc/arm/9.12/man.nsupdate.html)

dibdot, 2019-05-18:
ping @laurikimmel

dibdot, 2019-05-31:
no feedback by TO.

closed 2019-05-31 by dibdot

bluknight, 2021-05-16:
If we can re-open this, I'd be interested.  I did test the suggestion from @dibdot above with no success.  (at least from the LUCI interface -- I'm going to be testing in the code in a sec)

If needed, I'm happy to open a new issue.

bluknight, 2021-05-16:
Disregard -- my bad.  @dibdot's suggestion does work, with the caveat that _there is no space between the hash algorithm and the key name_.  therefore, a username setting of `hmac-256:keyname` would work, where `hmac-256: keyname` would not.

It would be a nice-to-have to be able to set the hash algorithm in the config programmatically (and also in the LUCI interface) but this is (at least to me) an acceptable workaround.