openwrt/packages #8930: Last nfs-kernel-server not work (2.3.3-2)

misterhsp, 2019-05-08:
The last nfs-kernel-server (2.3.3-2) update does not work, no more access to the share. A showmount -e results in 192.168.0.xxx 'clnt_create: RPC: Unable to send'.

diizzyy, 2019-05-08:
@tripolar @BKPepe

BKPepe, 2019-05-08:
According to the version of nfs-kernel-server you should have running OpenWrt 18.06.02 on your router, right? The thing is that nfs-kernel-server wasn't tested anyhow on that stable release, which is a pity. However a few days ago there was an issue, so I tried to fix it at least, but after that maintainer took a different approach and since that, I didn't look at it. To be honest, I lost my interest in nfs-kernel-server as there is also one issue, which I mention in my PR, but there wasn't done anything to it. I don't give up so easily and will try it again. I won't be able to look at it until next week.

misterhsp, 2019-05-08:
Yes right, 18.06.02. But it could also be due to rpcbind, but I am so sure.

...

Andy2244, 2019-05-10:
> The last nfs-kernel-server (2.3.3-2) update does not work, no more access to the share. A showmount -e results in 192.168.0.xxx 'clnt_create: RPC: Unable to send'.

Keep in mind `showmount -e` did not work, because rpcbind was build without `rmtcalls` support. This was fixed in snapshots 6 days ago.

However this did not prevent mounting the share, the share was just not browse-able.

misterhsp, 2019-05-10:
before the nfs-server update everything works fine. Now it doesn't work
anymore.

showmount -e xxxxxxx
clnt_create: RPC: Unable to send

mount -t nfs -o vers=3 xxxxxx:/data/halde /media/halde
mount.nfs: Network is unreachable

Tony-il-Capo, 2019-05-11:
This is probably the same problem me and other people are also having, as described [on the openwrt forum](https://forum.openwrt.org/t/sudden-nfs-failure-on-18-06-2-after-package-upgrade/36289)

A possible temporary fix is uninstalling latest nfs-kernel-version version and manually install older version from 17.xx.x repository, this is very easy, or you can try to compile the package yourself...

misterhsp, 2019-05-11:
Downgrade works, thanks.

...

tripolar, 2019-05-11:
try the latest version it should fix the problem, i removed rpcbind and nfs-kernel-server now depends on portmap

misterhsp, 2019-05-11:
The last version is still 2.3.3-2 in repo. Which version is the new one?

...

BKPepe, 2019-05-11:
Wow. Testing on users on stable releases. What a surprise. Can you please use pull requests if possible? It would be highly appreciated if you can try it locally if it works. Now it seems that you are just *blindly* commiting it directly to the repository.

Also, it takes time before all targets are built. Give it a time.

tripolar, 2019-05-11:
this was the proposed change and i thought it would be a better idea to use rpcbind which broke some stuff. I tested it and it worked here. Please test it with your config and report back if it works for you

Tony-il-Capo, 2019-05-11:
> The last version is still 2.3.3-2 in repo. Which version is the new one?

Wait for build-bot, should be there tomorrow, [v2.3.3-3 ](https://github.com/openwrt/packages/commit/54928eb69d60b6f05454732e1232a8de6669cb07)

misterhsp, 2019-05-13:
same error with v2.3.3-3 :(

...

Tony-il-Capo, 2019-05-13:
> same error with v2.3.3-3 :(

Yes, I tested it this morning and same problem, downgraded again.

misterhsp, 2019-05-14:
Pretty stupid a not working nfs-server, I hope it is fixed soon.

Tony-il-Capo, 2019-05-20:
Nope, and observing this whole situation I wouldn't count on it!.. 
Making changes to packages that no one or only himself tested, what the maintainer expected?! . And this is not limited to nfs-kernel-server package...

Better stick with old version for a while. Luckily it doesn't depend on specific kernel/unavailable packages... so it actually works...

tripolar, 2019-05-20:
how about providing patches? This is an open source project and not commercial support. You are free to participate instead.

tripolar, 2019-05-20:
how about providing patches? This is an open source project and not commercial support. You are free to participate.

davygrvy, 2019-06-07:
Any word on this.  I'm dead in the water

chunkeey, 2019-06-14:
Please see:

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=fcb41decf6c622482b20af45a77e62db8d95046e

https://forum.openwrt.org/t/nfs-broken-again/38020/24

for some (people with tiny 4MiB flash are left out) this might be fixed in the latest release.

misterhsp, 2019-11-07:
The problem is still present under 19.0.7.0-rc1 :(

...

Chris56, 2020-01-04:
See my post: https://github.com/openwrt/packages/issues/10942
Vanilla install of 18.06.5: Bang! Immideate errors and warnings, no nfsv3 anymore, breaks all former installtions using v3.
Prior to the release there even cannot have been a simple install! Let alone any serious testing.

nakamura-takashi, 2020-05-02:
I've succeeded at following setting.
server(openwrt 18.06.5)
root@LEDE:~# cat /etc/exports
/mnt/sda1    *(fsid=0,rw,insecure,sync,no_subtree_check)
root@LEDE:~#

client(ubuntu)
sudo /sbin/mount.nfs -v LEDE:/ /mnt/nfs_share

Tha-Fox, 2020-06-12:
> I've succeeded at following setting.
> server(openwrt 18.06.5)
> root@LEDE:~ cat /etc/exports
> /mnt/sda1 *(fsid=0,rw,insecure,sync,no_subtree_check)
> root@LEDE:~
> 
> client(ubuntu)
> sudo /sbin/mount.nfs -v LEDE:/ /mnt/nfs_share

I was also able to mount it that way. For me, in Arch Linux, the mount command was:

sudo /sbin/mount.nfs4 -v 192.168.0.1:/ /mnt/nfs_share

neheb, 2020-07-23:
What's the status on this?

plntyk, 2021-03-04:
Replicated it: NFS not working on 18.06.9  : tested x86 VM (via imagebuilder + qemu)

Imagebuilder was

    make image PACKAGES="nfs-kernel-server ss mtr"

Qemu cmdline

    qemu-system-i386 -k de -hda openwrt-18.06.9-x86-generic-combined-ext4.img  -netdev bridge,br=virbr0,id=hn0  -device e1000,netdev=hn0,id=nic1

in logread/dmesg:
svc: failed to register nfsdv3 RPC service (errno 97).
svc: failed to register lockdv1 RPC service (errno 97).

daemon.notice rpc.statd[1927]: Version 2.3.3 starting
daemon.warn rpc.statd[1927]: Flags: No-Daemon TI-RPC
daemon.notice rpc.statd[1927]: Initializing NSM state
daemon.warn rpc.statd[1927]: Running as root.  chown /var/lib/nfs to choose different user
daemon.err rpc.statd[1927]: failed to create RPC listeners, exiting

daemon.err rpc.mountd[1928]: Could not bind socket: (98) Address in use
daemon.err rpc.mountd[1928]: Could not bind socket: (98) Address in use
daemon.err rpc.mountd[1928]: Could not bind socket: (98) Address in use
daemon.err rpc.mountd[1928]: Could not bind socket: (98) Address in use
daemon.warn rpc.mountd[1928]: mountd: No V2 or V3 listeners created!
daemon.notice rpc.mountd[1928]: Version 2.3.3 starting

NFSD: Unable to end grace period: -110

rpcinfo output not-working vs. working

 rpcinfo -p 192.168.122.76
   program vers proto   port  service
    100000    2   tcp    111  portmapper
    100000    2   udp    111  portmapper
    100003    3   udp   2049  nfs
    100003    3   tcp   2049  nfs
    100003    4   tcp   2049  nfs
    100021    1   udp  32777  nlockmgr
    100021    3   udp  32777  nlockmgr
    100021    4   udp  32777  nlockmgr
    100021    1   tcp  32777  nlockmgr
    100021    3   tcp  32777  nlockmgr
    100021    4   tcp  32777  nlockmgr

rpcinfo -p 192.168.122.76
   program vers proto   port  service
    100000    4   tcp    111  portmapper
    100000    3   tcp    111  portmapper
    100000    2   tcp    111  portmapper
    100000    4   udp    111  portmapper
    100000    3   udp    111  portmapper
    100000    2   udp    111  portmapper
    100003    3   tcp   2049  nfs
    100003    4   tcp   2049  nfs
    100021    1   udp  32777  nlockmgr
    100021    3   udp  32777  nlockmgr
    100021    4   udp  32777  nlockmgr
    100021    1   tcp  32777  nlockmgr
    100021    3   tcp  32777  nlockmgr
    100021    4   tcp  32777  nlockmgr
    100005    1   udp  32780  mountd
    100005    1   tcp  32780  mountd
    100005    2   udp  32780  mountd
    100005    2   tcp  32780  mountd
    100005    3   udp  32780  mountd
    100005    3   tcp  32780  mountd
    100024    1   udp  32778  status
    100024    1   tcp  32778  status

Even with empty return of "lsof -i" manual startup of rpc.mountd according to initfile (/usr/sbin/rpc.mountd -p 32780 ) fails with the could not bind socket error

rpcinfo -e <ip of openwrt> results in missing mountd entries, showexport fails, NFS mount does not work

21.02-SNAPSHOT is working by default - tested on x86-64 VM via imagebuilder with same make image command that has 18.06 not working

19.09.7 is working too

plntyk, 2021-03-04:
fixing could be either testing newer nfs-kernel-server versions or running down some strace rabbit hole as rpc.mountd does not seem to provide much debug output when run in foreground

closed 2025-11-14 by GeorgeSapkin