openwrt/packages #8968: netdata: log spam

anomeome, 2019-05-12:
On target mvebu with current master image, the log is (intermittently) being spammed every 2 secs with:

`Sun May 12 14:08:18 2019 daemon.err netdata[3760]: PROCFILE: Cannot open file '/proc/sysvipc/shm'`

unclear when this started, or what it takes to initiate the spam.

Maintainer: @diizzyy @BKPepe

BKPepe, 2019-05-12:
Are you using default configuration or not? Because I've just installed netdata again on my router and I don't have your issue.

Anyway, I'd suggest you disable plugin:proc:ipc to avoid spamming your syslog. This can be done once you download the latest netdata configuration file - http://IP_where_netdata_is_running:19999/netdata.conf , edit the file and replace the current one in /etc/netdata/netdata.conf.

anomeome, 2019-05-12:
k, thanks. The conf file would be the original OOTB as I have never changed the config. I wonder if this is related to _kernel namespaces ipc_.

closed 2019-05-12 by anomeome