openwrt/packages #8974: sysntpd not listening on ifaces

n8v8R, 2019-05-13:
Maintainer: @tripolar 

Environment: 

- package feed from OpenWRT 18.06
- DHCP ipv4/ipv6 provided by `odhcpd` (`dnsmasq` disabled)

`ubus -S call system board`
{"kernel":"4.14.113","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"armada-385-turris-omnia","release":{"distribution":"TurrisOS","version":"4.0-beta1","revision":"0663455801","target":"mvebu\/cortexa9","description":"TurrisOS 4.0-beta1 0663455801"}}

`ubus list network.interface.*`
> network.interface.guest
> network.interface.lan
> network.interface.loopback
> network.interface.mgt
> network.interface.wan
> network.interface.wg0
___
Description: With he below settings the server is only listening on the lo iface but not others though my understanding is that if `interface` parameter is omitted it defaults to all interfaces.
```
config timeserver 'ntp'
	option enabled '1'
	option enable_server '1'
```
Yet `ss -tulpn | grep ntp` produces only

> udp  UNCONN 0      0                         127.0.0.1:123         0.0.0.0:*     users:(("ntpd",pid=24507,fd=48))
> udp  UNCONN 0      0                             [::1]:123               *:*     users:(("ntpd",pid=24507,fd=49))

Adding `list interface` parameters does not make a difference.

tripolar, 2019-05-13:
I'm not the maintainer of sysntpd, i maintain ntpd

tripolar, 2019-05-13:
I'm not the maintainer of sysntpd, i maintain ntpd
With ntpd, the server listens on all interfaces when no interface is specified

n8v8R, 2019-05-13:
@tripolar oops, my apology that I got that mixed up :flushed: 

Unfortunately `ntpd` does (currently) not install on TOS https://gitlab.labs.nic.cz/turris/turris-os-packages/issues/289

One question though about `ntpd` - in the LEDE trunk it used to be highly configurable via /et/ntp.conf which however is not present in 18.06/Master - is `ntpd ` still configurable that way or limited to /etc/config/system?

Tony-il-Capo, 2019-05-15:
> One question though about `ntpd` - in the LEDE trunk it used to be highly configurable via /et/ntp.conf which however is not present in 18.06/Master - is `ntpd ` still configurable that way or limited to /etc/config/system?

Limited...
See [here](https://github.com/openwrt/packages/issues/8992)

closed 2019-05-16 by n8v8R