openwrt/packages #8977: libre2: missing dependency when being used by an example package

James-TR, 2019-05-14:
Maintainer: @ammubhave 
Environment: ARM - WRT1900ACS - Trunk

Description:
I am in the process of creating a new Makefile for DNSDIST https://dnsdist.org and when trying to use the recently added libre2, I'm receiving the following error despite everything looking to be ok:

```
Package dnsdist is missing dependencies for the following libraries:
libre2.so
```

I've tried messing around with the libre2 Makefile but don't know what needs to be changed to have the include work. For reference my Makefile for DNSDIST is below. I'm still working on getting it finished and still have some other dependencies to add in :)

```
include $(TOPDIR)/rules.mk

PKG_NAME:=dnsdist
PKG_VERSION:=1.4.0-alpha2
PKG_RELEASE:=1

PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)-$(PKG_VERSION)
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.bz2
PKG_SOURCE_URL:=https://downloads.powerdns.com/releases/
PKG_HASH:=9159e9ce02dfa49b509e9c1a89e672a3ede7f9e60639ed40e3135b0eb29515c3

PKG_LICENCE:=GPLv2
PKG_LICENCE_FILES:=COPYING

include $(INCLUDE_DIR)/package.mk

PKG_BUILD_DEPENDS:=boost

define Package/dnsdist
  SECTION:=base
  CATEGORY:=Network
  TITLE:=dnsdist DNS-, DOS- and abuse-aware loadbalancer.
  DEPENDS:=+protobuf +re2 +libedit +libsodium +lua +boost +libopenssl
  URL:=http://dnsdist.org/
endef

define Package/dnsdist/description
endef

define Build/Configure
        $(call Build/Configure/Default, \
                --prefix=/usr \
                --sysconfdir=/etc \
                --with-lua \
                --enable-dnscrypt \
                --with-libsodium \
                --with-protobuf \
                --with-re2 \
                --enable-dns-over-tls \
          )
endef

define Package/dnsdist/install
        $(INSTALL_DIR) $(1)/usr/bin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/dnsdist $(1)/usr/bin/
endef

$(eval $(call BuildPackage,dnsdist))
```

ammubhave, 2019-05-14:
That is strange. Just to be sure, you've run `make menuconfig`, and selected the dnsdist package (as a result of which, the re2 package should also have gotten selected)?

James-TR, 2019-05-14:
Yeah the dnsdist package has been selected in menuconfig and re2 was included in line with the dependencies. I initially also had issues with some openssl stuff missing with the same error but that was fixed by adding +libopenssl.

I'm not sure of there is something with the build system where it tracks the provided libraries?

James-TR, 2019-05-14:
I was able to get it to build correctly by adding this to the Makefile:
```
define Build/InstallDev
       $(INSTALL_DIR)                                          $(1)/usr/include/re2
       $(CP) $(PKG_INSTALL_DIR)/usr/include/re2/*.h            $(1)/usr/include/re2/

       $(INSTALL_DIR)                                          $(1)/usr/lib
       $(CP) $(PKG_INSTALL_DIR)/usr/lib/libre2.so              $(1)/usr/lib/
endef
```

I'm running tests on a clean repo to ensure this is what's needed to fix it, as it may have been a fluke!

James-TR, 2019-05-19:
Unable to reproduce on another fresh pull of the SDK. After further investigation there may have been other issues with my copy of the SDK that mad it seem like there were problems.

closed 2019-05-19 by James-TR