openwrt/packages #8978: python-curl: fails to compile if libcurl uses libopenssl

val-kulkov, 2019-05-14:
Maintainer: @valdi74 
Attention: @BKPepe 
Environment: bcm53xx, Buffalo WXR-1900DHP, OpenWrt SNAPSHOT r9992-86fd8cb

Description:

Since commit b3d1401dccc3548aa8efec331817a2521797e230 (PR #8325), python-curl fails to compile if libcurl uses openssl as its SSL library rather than mbedtls:
```
Package python-curl is missing dependencies for the following libraries:
libmbedtls.so.12
Makefile:62: recipe for target '/home/val/Development/openwrt/targets/x86_64/bin/packages/x86_64/packages/python-curl_7.43.0.2-2_x86_64.ipk' failed
make[3]: *** [/home/val/Development/openwrt/targets/x86_64/bin/packages/x86_64/packages/python-curl_7.43.0.2-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/val/Development/openwrt/targets/x86_64/feeds/packages/lang/python/python-curl'
time: package/feeds/packages/python-curl/python/compile#2.62#0.20#3.75
package/Makefile:107: recipe for target 'package/feeds/packages/python-curl/compile' failed
make[2]: *** [package/feeds/packages/python-curl/compile] Error 2
```

Adding the dependency on libmbedtls in python-curl's Makefile fixes the compilation error, but IMO in a wrong way. [Hardcoding the use of mbedtls in python-curl's Makefile](https://github.com/BKPepe/packages/blob/b3d1401dccc3548aa8efec331817a2521797e230/lang/python/python-curl/Makefile#L58-L59) without regard to the choice of SSL library for curl is wrong. The choice of the SSL library should be based on libcurl's SSL library configuration setting.

BKPepe, 2019-05-14:
Yeah, it fails, but I think it's better if it fails than broken package as it was. I explained it in that PR, which you mention.

diizzyy, 2019-05-14:
Given that (lib)curl is a dependency for python-curl I guess you could test which dependency of ssl library libcurl is using?

neheb, 2019-07-02:
https://github.com/openwrt/packages/pull/8979 was merged. Closing.

closed 2019-07-02 by neheb