openwrt/packages #8985: gammu: not found

bruceleerabbit, 2019-05-14:
Maintainer: @aTanW 
Environment:
$ uname -r
4.14.95

Description:
I'm looking to install gammu.  When I run `opkg list | grep gammu` there is nothing.  Is this caused by the [migration](https://github.com/openwrt/packages/issues/8893) to python 3?  Someone mentioned having [trouble](https://github.com/openwrt/packages/issues/1522) building it.  I don't even see how to build it.  The Gammu [installation guide](https://github.com/gammu/gammu/blob/master/docs/manual/project/install.rst) omits openwrt.

neheb, 2019-05-14:
which platform?

neheb, 2019-05-14:
I see it here: https://downloads.openwrt.org/releases/18.06.2/packages/mipsel_74kc/packages/gammu_1.38.4-2_mipsel_74kc.ipk

bruceleerabbit, 2019-05-14:
Oh, sorry for the trouble.. apparently I needed do an `opkg update`.

closed 2019-05-14 by bruceleerabbit