openwrt/packages #8989: gammu: No response in specified timeout. Probably phone not connected.

bruceleerabbit, 2019-05-15:
Maintainer: @aTanW or @neheb or @micmac1
Environment:
$ uname -r
4.14.95
openwrt install image is: `openwrt-18.06.2-brcm47xx-mips74k-asus-rt-n16-squashfs.trx`

## the problem
$ `gammu -c /root/gammu.rc getallsms`
No response in specified timeout. Probably phone not connected.

## background info
I have gammu installed on both my Debian pc and on openwrt.  There are no problems on Debian with my device (huawei e220).  There were no errors from running `opkg install gammu`.  Configuring it is a bit rough-going.  The `gammu-detect` and `gammu-config` commands are not found on openwrt.  So I configured manually using the working Debian config as a reference.

There is a `/etc/config/gammu` file:
```
config daemon default
	option enabled		0
	option device		/dev/ttyUSB0
	option connection	at
	option logfile		syslog
```
The first possible issue is that there is no existing `/dev/ttyUSB0`.  I left it untouched initially just in case that names a device that will be created.  The next problem is that this file is strangely formatted.  It does not conform to the format that `gammu` expects and so running `gammu -c /etc/config/gammu` results in a rejection of that config file.  So then I created `/root/gammu.rc`:
```
[gammu]
port=/dev/ttyS0
connection=at
name=huawei E220
```
Note that although the working port on debian is `/dev/ttyUSB0`, it's apparently mapped differently in openwrt.  The `/dev/ttyS0` comes from running: `(dmesg; logread) | grep -iE '(usb|tty)'` which shows:
```
[    0.000000] Kernel command line: noinitrd console=ttyS0,115200
[    0.000000] bcma: bus0: Core 4 found: USB 2.0 Host (manuf 0x4BF, id 0x819, rev 0x04, class 0x0)
[    0.677837] console [ttyS0] disabled
[    0.698082] serial8250.0: ttyS0 at MMIO 0xb8000300 (irq = 2, base_baud = 1250000) is a U6_16550A
[    1.345591] console [ttyS0] enabled
[    4.063457] usbcore: registered new interface driver usbfs
[    4.069167] usbcore: registered new interface driver hub
[    4.074848] usbcore: registered new device driver usb
[    4.088697] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    4.109529] ehci-platform ehci-platform.0: new USB bus registered, assigned bus number 1
[    4.151496] ehci-platform ehci-platform.0: USB 2.0 started, EHCI 1.00
[    4.159589] hub 1-0:1.0: USB hub found
[    4.173865] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    4.194703] ohci-platform ohci-platform.0: new USB bus registered, assigned bus number 2
[    4.277002] hub 2-0:1.0: USB hub found
[   12.541545] usb 2-1: new full-speed USB device number 2 using ohci-platform
[   14.356268] usbcore: registered new interface driver btusb
[   14.569649] Bluetooth: RFCOMM TTY layer initialized
[..]kern.info kernel: [    0.677837] console [ttyS0] disabled
[..]kern.info kernel: [    0.698082] serial8250.0: ttyS0 at MMIO 0xb8000300 (irq = 2, base_baud = 1250000) is a U6_16550A
[..]kern.info kernel: [    1.345591] console [ttyS0] enabled
[..]kern.info kernel: [    4.063457] usbcore: registered new interface driver usbfs
[..]kern.info kernel: [    4.069167] usbcore: registered new interface driver hub
[..]kern.info kernel: [    4.074848] usbcore: registered new device driver usb
[..]kern.info kernel: [    4.088697] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[..]kern.info kernel: [    4.109529] ehci-platform ehci-platform.0: new USB bus registered, assigned bus number 1
[..]kern.info kernel: [    4.151496] ehci-platform ehci-platform.0: USB 2.0 started, EHCI 1.00
[..]kern.info kernel: [    4.159589] hub 1-0:1.0: USB hub found
[..]kern.info kernel: [    4.173865] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[..]kern.info kernel: [    4.194703] ohci-platform ohci-platform.0: new USB bus registered, assigned bus number 2
[..]kern.info kernel: [    4.277002] hub 2-0:1.0: USB hub found
[..]kern.info kernel: [   12.541545] usb 2-1: new full-speed USB device number 2 using ohci-platform
[..]kern.info kernel: [   14.356268] usbcore: registered new interface driver btusb
[..]kern.info kernel: [   14.569649] Bluetooth: RFCOMM TTY layer initialized
```
There is only mention of `ttyS0` and no mention of `ttyusb0`.

Running `gammu -c /root/gammu.rc getallsms` gives "No response in specified timeout. Probably phone not connected."

So then I edited `/etc/config/gammu` to make the device reference `/dev/ttyS0` (so both `/etc/config/gammu` and `/root/gammu.rc` are consistent).  This still makes no difference even after a reboot.

micmac1, 2019-05-15:
Hi all,

Please take me off these notifications.

Thanks,
Seb

aTanW, 2019-05-21:
@micmac Please, press "Unsubscribe" button on thre right side of this page.

aTanW, 2019-05-21:
@bruceleerabbit Seems, that your dongle does not recognized as tty device. Try to search for your model name together with "openwrt" keyword.

neheb, 2020-01-24:
Is this still an issue?

neheb, 2020-02-23:
I'm closing this.

/etc/config/gammu is meant to be used by the init script, not directly. If the device path is wrong, fix it.

closed 2020-02-23 by neheb