openwrt/packages #8992: ntpd: init.d script, use standard config file (/etc/ntp.conf) if exists

Tony-il-Capo, 2019-05-15:
Maintainer: @tripolar 
Environment: (all, OpenWrt 18.06.2)

I think it would be better if ntpd would use standard configuration file (/etc/ntp.conf) if file exists instead of always generating/reading values from /etc/config/system.
ntpd no longer installs /etc/ntp.conf by default, so if user have that file there he wants that config to be used.

Using UCI to configure ntpd can be frustrating, also I'm not sure if current script reads all possible parameters from /etc/config/system, it may also 'emit' and ignore (unwanted) flags, that's why I think would be better to prioritize standard configuration file for advance usage.

jonathanunderwood, 2019-05-15:
Many services have a toggle option in the UCI config that allows a user to specify that the native config file should be used directly. See stubby for an example.

tripolar, 2019-05-15:
i see no point in providing this "feature" as it just confuses the user

closed 2019-05-15 by tripolar

Tony-il-Capo, 2019-05-15:
Really? How?
'Basic' user will not be even aware of this, a simple 'if' check in the script would do the job.

Without it people are manually editing init.d script and deleting  <-c path-to-config> argument , because without it, ntpd defaults to /etc/ntp.conf, so it uses their custom configuration.

I posted this because I had several people asking for help re-configuring their ntpd server for their LANs, after I advised them to upgrade to last OpenWrt release, and the problem was down to this. For me, you can leave as 'you' like, just beware that it doesn't work as good as you think. 

Also, [readme](https://github.com/openwrt/packages/blob/master/net/ntpd/README.md) say nothing about 'restrict' statements (how a 'user' can override them/add another), although the script always passes them to ntpd config.

jonathanunderwood, 2019-05-16:
Hi Tony. I happen to agree with you. But perhaps the maintained would be more inclined to engage if you made a PR implementing the change, so the discussion is about the implementation - questions of simplicity or confusion are best had about something tangible.

Many packages already do as you suggest, by the way, but through an explicit option on the uci config file.

Matux95, 2019-12-06:
Don't forget Squar readers my friends