openwrt/packages #9005: <mosquitto-ssl-1.6.2>:  preprocessor compatibility issue

d51r3verse, 2019-05-16:
/w CONFIG_GCC_USE_VERSION_8
```
util_mosq.c:32:43: error: missing binary operator before token "("
 #  if defined(__GLIBC__) && __GLIBC_PREREQ(2, 25)
                                           ^
```

I'm not sure this is the proper mitigations
```
--- a/lib/util_mosq.c   2019-04-30 22:34:20.000000000 +0900
+++ b/lib/util_mosq.c   2019-05-16 18:45:37.362140999 +0900
@@ -29,8 +29,10 @@
 #endif

 #if !defined(WITH_TLS) && defined(__linux__)
-#  if defined(__GLIBC__) && __GLIBC_PREREQ(2, 25)
-#    include <sys/random.h>
+#  if defined(__GLIBC__)
+#    if __GLIBC_PREREQ(2, 25)
+#      include <sys/random.h>
+#    endif
 #  endif
 #endif

@@ -325,10 +327,12 @@
        if(RAND_bytes(bytes, count) == 1){
                rc = MOSQ_ERR_SUCCESS;
        }
-#elif defined(__GLIBC__) && __GLIBC_PREREQ(2, 25)
+#elif defined(__GLIBC__)
+#if __GLIBC_PREREQ(2, 25)
        if(getrandom(bytes, count, 0) == 0){
                rc = MOSQ_ERR_SUCCESS;
        }
+#endif
 #elif defined(WIN32)
        HRYPTPROV provider;

```

karlp, 2019-05-16:
reported to upstream, who says this looks reasonable.  I'll patch this in our tree next week when I'm back in town.

lynxis, 2019-05-22:
@kalp which upstream patch do you mean? Do you have a link?

karlp, 2019-05-22:
https://github.com/eclipse/mosquitto/commit/d05bd958813b7ead344f66653ce9b13b00e67595

There's quite a few bugs shaking out of the new 1.6 branch, I'm expecting a new dot release anyday, so I haven't really felt like trying to patch every single one of them as they land.  IF gcc 8 was the default I'd be more inclined to patch it immediately.

karlp, 2019-06-19:
1.6.3 now available in master via 08ab75d88 and 19.07 via 03e1633d1

closed 2019-06-19 by karlp