openwrt/packages #9022: libfstrm: missing URL and TITLE is too long

BKPepe, 2019-05-17:
Maintainer: @James-TR

Description:

Recently added package #8983 has formal issues.

- Missing URL
- PKG_TITLE is too long and that's why the title isn't shown in make menuconfig
[image]

Not sure, but isn't there missing dependency, which has @ja-pa in his [Makefile](https://gitlab.labs.nic.cz/turris/turris-os-packages/commit/953d0ed46e9c0769e6ed53b1759bde2984356c71) in his dev branch?

James-TR, 2019-05-18:
Hello, I am not aware of any missing dependencies for this package and have confirmed it works correctly already. PR #9027 should resolve the remaining issues with formatting and even fixed up the description as it did not show correctly.

James-TR, 2019-05-18:
The dependency for libevent2 is not required as it is only used to build the example programs provided by the package, which are disabled by passing the "--disable-programs" CONFIGURE flag.

As addressed in the fstrm github:
> Reference programs fstrm_capture, fstrm_dump, and fstrm_replay are provided. In order to build fstrm_capture, the libevent library must be installed. The option --disable-programs can be passed to configure to disable building these programs.

PR #9027 (2019-05-19): libfstrm: fix formality issues
Maintainer: me

Correct issues with TITLE and add URL pointing to the fstrm GitHub Page. Reformat the description so it renders correctly in menuconfig.

Fixes #9022
commit edcb8787 (2019-05-19): libfstrm: fix formality issues

Maintainer: me

Correct issues with TITLE and add URL pointing to the fstrm GitHub Page. Reformat the description so it renders correctly in menuconfig.

Fixes #9022
closed 2019-05-19 by neheb