openwrt/packages #9054: ddns-scripts: fails due to race condition when getting local IPv6

patrakov, 2019-05-23:
Maintainer: @chris5560
Environment: mips_24kc, Buffalo WBMR-HP-G300H, OpenWRT 18.01

Description:

I have configured ddns-scripts to update my dynv6 hostname with the IPv6 address of the LAN, like this:

```
config service 'dynv6_ipv6'
	option use_ipv6 '1'
	option service_name 'dynv6.com'
	option enabled '1'
	option lookup_host 'hostname.dynv6.net'
	option domain 'hostname.dynv6.net'
	option password 'password'
	option use_https '1'
	option interface 'lan'
	option ip_network 'lan'
	option force_interval '1'
```

My ISP changes the IPv6 block allocated to my router daily (forces PPPoE disconnect). And the ddns scripts on the router never recover after that:

```
 135711       : ************ ************** ************** **************
 135715  note : PID '1661' started at 2019-05-15 13:57
 135715       : ddns version  : 2.7.8-1
 135716       : uci configuration:
ddns.dynv6_ipv6.domain='hostname.dynv6.net'
ddns.dynv6_ipv6.enabled='1'
ddns.dynv6_ipv6.force_interval='1'
ddns.dynv6_ipv6.interface='lan'
ddns.dynv6_ipv6.ip_network='lan'
ddns.dynv6_ipv6.lookup_host='hostname.dynv6.net'
ddns.dynv6_ipv6.password='***PW***'
ddns.dynv6_ipv6.service_name='dynv6.com'
ddns.dynv6_ipv6.use_https='1'
ddns.dynv6_ipv6.use_ipv6='1'
ddns.dynv6_ipv6=service
 135721       : verbose mode  : 0 - run normal, NO console output
 135722       : check interval: 600 seconds
 135723       : force interval: 3600 seconds
 135723       : retry interval: 60 seconds
 135724       : retry counter : 0 times
 135724       : No old process
 135725       : last update: never
 135725       : Detect registered/public IP
 135726       : #> /usr/bin/nslookup hostname.dynv6.net  >/var/run/ddns/dynv6_ipv6.dat 2>/var/run/ddns/dynv6_ipv6.err
 135732 ERROR : BusyBox nslookup error: '1'
 135733       : ;; connection timed out; no servers could be reached
 135733       : Detect registered/public IP
 135734       : #> /usr/bin/nslookup hostname.dynv6.net  >/var/run/ddns/dynv6_ipv6.dat 2>/var/run/ddns/dynv6_ipv6.err
 135739       : Registered IP '2a01:540:a6f:3c00::1' detected
 132850  info : Starting main loop at 2019-05-22 13:28
 132852       : Detect local IP on 'network'
 132853 ERROR : Can not detect local IP using network_get_ipaddr6 'lan' - Error: '1' - TERMINATE
 132856  WARN : PID '1661' exit WITH ERROR '1' at 2019-05-22 13:28
```

I think this is a race condition - the old IPv6 address has already been removed, while the DHCPv6 conversation to obtain the new delegated prefix is in progress. Restarting the ddns service successfully updates the IP.

patrakov, 2019-05-23:
Actually it's different. See /etc/hotplug.d/iface/95-ddns:

```
#!/bin/sh
case "$ACTION" in
ifup)
/etc/init.d/ddns enabled && /usr/lib/ddns/dynamic_dns_updater.sh -n "$INTERFACE" -- start
;;
ifdown)
/usr/lib/ddns/dynamic_dns_updater.sh -n "$INTERFACE" -- stop
;;
esac
```

The problem is that the `ifup` event happens on LAN while WAN is not connected, and so there is no delegated IPv6 prefix at that time. And, when WAN connects, then there is nothing which would restart the updater on LAN.

GeorgeSapkin, 2025-11-14:
@patrakov is this still relevant?