openwrt/packages #9065: ULOGD2 version bump

CanadianJeff, 2019-05-24:
ULOGD2 is now version 2.0.7 but openwrt still shows version 2.0.5-2?

OpenWRT/LEDE Version???
http://downloads.openwrt.org/releases/18.06.2/packages/x86_64/packages/ulogd_2.0.5-2_x86_64.ipk

Updated Version!!!
https://git.netfilter.org/ulogd2/

can we have 2.0.7 in OpenWRT please?

neheb, 2019-05-24:
Latest snapshot has the latest version.

CanadianJeff, 2019-05-24:
hrmmmmmmmm I just check and see 2.0.7
https://github.com/openwrt/packages/blob/master/net/ulogd/Makefile

but its not in the package repos?

neheb, 2019-05-24:
You linked the 18.06.2. That's not snapshot.

CanadianJeff, 2019-05-24:
oh yes I see it now
http://downloads.openwrt.org/snapshots/packages/x86_64/packages/ulogd_2.0.7-1_x86_64.ipk

but why can we not have this in 18.06.2 repos?

CanadianJeff, 2019-05-24:
tried to add snapshots to customfeed and got errors

```
root@APU2:/tmp# nano /etc/opkg/customfeeds.conf
root@APU2:/tmp# opkg update
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_snapshot
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.1/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/releases/18.06.1/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/telephony/Packages.sig
Signature check passed.
root@APU2:/tmp# opkg install ulogd
Upgrading ulogd on root from 2.0.5-2 to 2.0.7-1...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/ulogd_2.0.7-1_x86_64.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ulogd:
 *      libmnl0 *       libnfnetlink0 *
 * opkg_install_cmd: Cannot install package ulogd.
root@APU2:/tmp#
```

not sure I want to break my install

neheb, 2019-05-24:
Generally speaking, the rule for stable branches is to backport security and bug fixes, not so much bump package versions. This does happen but on occasion.

ping @commodo since he last updated it.

CanadianJeff, 2019-05-25:
2.0.7 offers remote json logging and 2.0.5 is over 4 years old now??????

commodo, 2019-05-25:
hey

(i usually don't reply to messages during the weekend, but every-now-n-then i have some time to do some replying)

so, i would second what @neheb said about not adding this to 18.06;  this policy should be enforced as much as possible; if we allow exceptions, it becomes a "pandora's box" ;
in any case, i think there will be a 19.x release , which should contain this new version;

as I recall, @psycho-nico is/was not very active for a while;
not sure if that changed recently ?

i can take over that package as well (as maintainer)
i am swamped with other stuff (in the meantime), so I cannot alloc too much time for OpenWrt packages these days;
hopefully things will clear in a few weeks for me (as I have plenty of backlog here)

neheb, 2019-07-02:
@CanadianJeff OpenWrt 19.07 will be out soon.

CanadianJeff, 2019-07-09:
how soon?

neheb, 2019-07-09:
19.07 has been branched. I'd say less than a month.

CanadianJeff, 2019-07-25:
ok so I have given SNAPSHOT a try
```
root@APU2:/# cat /etc/banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r10513-4a06d62f0e
 -----------------------------------------------------
```
trying to get the remote json logging setup
```
stack=log1:NFLOG,base1:BASE,ifi1:IFINDEX,ip2str1:IP2STR,mac2str1:HWHDR,json1:JSON
```
```
root@APU2:/# cat /etc/ulogd.conf | grep -A8 '\[json1\]'
[json1]
sync=1
file="/var/log/ulogd.json"
timestamp=0
device="APU2-SHAWSECURE"
boolean_label=1
mode=tcp
host=192.168.2.10
port=10210
```
the ulogd.json file gets created but it wont send any data to remote server (checked with netstat and tcpdump)

neheb, 2019-07-25:
Did you port forward?

CanadianJeff, 2019-07-25:
I doubt port forwarding is needed here because the target server is on the same subnet/lan as the router

CanadianJeff, 2019-07-27:
could we have this fixed or looked at before openwrt 19.04 so I can deploy this setup to over 200 routers?

neheb, 2019-07-27:
From a quick look, it sounds like you want this backported: https://git.netfilter.org/ulogd2/commit/?id=9d9ea2cd70a369a7f665a322e6c53631e01a2570

CanadianJeff, 2019-07-28:
yup

closed 2019-07-31 by neheb

CanadianJeff, 2019-08-06:
SEGFAULTS NOW?

```
[  336.017775] ulogd[5403]: segfault at 7f3707b21048 ip 0000000000405265 sp 00007ffed6238ba0 error 4 in ulogd[403000+4000]
[  336.028696] Code: 00 00 31 c0 e8 7e ed ff ff eb 05 48 8b 1b eb 90 49 8b 44 24 28 48 8b 68 10 49 8b 44 24 28 48 83 c0 10 48 39 c5 74 41 45 31 ed <44> 3b 6d 68 73 32 44 89 eb 48 6b db 48 48 03 5d 60 66 8b 43 06 a8

BusyBox v1.31.0 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r10641-1d2b2e744e
 -----------------------------------------------------
root@APU2:/#
```
root@APU2:/tmp/log# tail ulogd.log
Mon Aug  5 19:31:34 2019 <1> ulogd.c:890 assigning `oob.protocol(?)' as source for BASE(oob.protocol)
Mon Aug  5 19:31:34 2019 <1> ulogd.c:827 traversing plugin `NFLOG'
Mon Aug  5 19:31:34 2019 <3> ulogd.c:912 log1 instance already loaded
Mon Aug  5 19:31:34 2019 <1> ulogd_output_JSON.c:186 connecting to 192.168.2.10:10210
Mon Aug  5 19:31:34 2019 <7> ulogd.c:947 error starting `json1'
Mon Aug  5 19:31:34 2019 <1> ulogd.c:1040 destroying stack
Mon Aug  5 19:31:34 2019 <3> ulogd.c:1647 initialization finished, entering main loop
Mon Aug  5 19:36:46 2019 <7> ulogd_output_JSON.c:249 Failure sending message: Function not implemented
Mon Aug  5 19:36:46 2019 <1> ulogd_output_JSON.c:186 connecting to 192.168.2.10:10210
Mon Aug  5 19:36:46 2019 <5> ulogd.c:621 error during propagate_results
root@APU2:/tmp/log# opkg info ulogd
Package: ulogd
Version: 2.0.7-3
Depends: libc, libmnl0, libnfnetlink0, libpthread
Status: install user installed
Architecture: x86_64
Conffiles:
 /etc/ulogd.conf 9af3c360c2912b839f970272f9f2e0ddcb74a0994b1bf4fa4e1e5e6b840adc62
Installed-Time: 1565058612

root@APU2:/tmp/log#
```

CanadianJeff, 2019-08-06:
segfault does not happen when I comment out the TCP MODE

```
[json1]
sync=1
file="/var/log/ulogd.json"
timestamp=0
device="APU2-SHAWSECURE"
boolean_label=1
#mode=tcp
#host=192.168.2.10
#port=10210
```

neheb, 2019-08-06:
I lied. Something is returning -ENOSYS. Maybe a kernel configuration issue.

CanadianJeff, 2019-08-06:
so should we report this upstream?

neheb, 2019-08-06:
Why not.