openwrt/packages #9066: <python-pycparser>: build error

brucealthompson, 2019-05-24:
Maintainer: Jeffery To <jeffery.to@gmail.com>
Environment: (X86, model, Github trunk)

Description:
The package python-pycparser does not build in the latest OpenWrt source tree in github as well as the latest OpenWrt SDK (https://downloads.openwrt.org/snapshots/targets/x86/generic/openwrt-sdk-x86-generic_gcc-7.4.0_musl.Linux-x86_64.tar.xz)
 
The basic problem is that the copy of python and python3 that ends up getting installed in staging_dir/hostpkg/bin does not have the python ply package installed. I am not sure why this is. I assume it is a bug in the latest OpenWrt source tree. I worked around the problem by installing the python ply package in staging_dir/hostpkg/lib/python{2.7,3.7}/site-packages
 
I have checked my host environment and the python ply package is installed in my host copies of python2.7 and python 3. This is why I assume this is a problem with the latest OpenWrt source tree and not with my OpenWrt cross compile environment.

Here's the output of make package/python-pycparser/compile:
```
echo "libc" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/package/libs/toolchain'
time: package/libs/toolchain/compile#0.02#0.01#0.13
make[2]: Entering directory '/home/brucet/buildbot/source/package/libs/zlib'
echo "zlib" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/package/libs/zlib'
time: package/libs/zlib/compile#0.02#0.00#0.12
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/libxml2'
echo "libxml2" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/libxml2.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/libxml2'
time: package/feeds/packages/libxml2/compile#0.04#0.02#0.16
make[2]: Entering directory '/home/brucet/buildbot/source/package/libs/uclibc++'
make[2]: Leaving directory '/home/brucet/buildbot/source/package/libs/uclibc++'
time: package/libs/uclibc++/compile#0.03#0.00#0.12
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/db47'
echo "libdb47" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/db47.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/db47'
time: package/feeds/packages/db47/compile#0.05#0.02#0.14
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/expat'
echo "libexpat" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/expat.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/expat'
time: package/feeds/packages/expat/compile#0.04#0.00#0.15
make[2]: Entering directory '/home/brucet/buildbot/source/package/libs/gettext-full'
make[2]: Leaving directory '/home/brucet/buildbot/source/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#0.04#0.02#0.16
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/gdbm'
echo "libgdbm" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/gdbm.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/gdbm'
time: package/feeds/packages/gdbm/compile#0.02#0.00#0.13
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/libffi'
echo "libffi" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/libffi.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/compile#0.03#0.00#0.14
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/lang/python/python-pip-conf'
echo "python-pip-conf" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/python-pip-conf.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/lang/python/python-pip-conf'
time: package/feeds/packages/python-pip-conf/compile#0.02#0.01#0.12
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/expat'
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/expat'
time: package/feeds/packages/expat/host-compile#0.02#0.01#0.13
make[2]: Entering directory '/home/brucet/buildbot/source/package/utils/bzip2'
make[2]: Leaving directory '/home/brucet/buildbot/source/package/utils/bzip2'
time: package/utils/bzip2/host-compile#0.03#0.00#0.14
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/lang/python/python'
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/lang/python/python'
time: package/feeds/packages/python/host-compile#0.08#0.02#0.18
make[2]: Entering directory '/home/brucet/buildbot/source/feeds/packages/libs/sqlite3'
echo "libsqlite3" >> /home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/pkginfo/sqlite3.default.install
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/libs/sqlite3'
time: package/feeds/packages/sqlite3/compile#0.03#0.00#0.12
make[2]: Entering directory '/home/brucet/buildbot/source/package/libs/ncurses'
make[2]: Leaving directory '/home/brucet/buildbot/source/package/libs/ncurses'
time: package/libs/ncurses/host-compile#0.04#0.00#0.14
make[2]: Entering directory '/home/brucet/buildbot/source/package/libs/ncurses'
[... 94 lines cut ...]
    self.run_command(cmd)
  File "/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "./setup.py", line 25, in run
    msg="Build the lexing/parsing tables")
  File "/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/usr/lib/python2.7/distutils/cmd.py", line 349, in execute
    util.execute(func, args, msg, dry_run=self.dry_run)
  File "/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/usr/lib/python2.7/distutils/util.py", line 314, in execute
    func(*args)
  File "./setup.py", line 18, in _run_build_tables
    cwd=os.path.join(dir, 'pycparser'))
  File "/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/brucet/buildbot/source/staging_dir/hostpkg/bin/python2.7', '-B', '_build_tables.py']' returned non-zero exit status 1
Makefile:70: recipe for target '/home/brucet/buildbot/source/build_dir/target-mips_mips32_musl/python-pycparser-2.19/.built' failed
make[2]: *** [/home/brucet/buildbot/source/build_dir/target-mips_mips32_musl/python-pycparser-2.19/.built] Error 1
make[2]: Leaving directory '/home/brucet/buildbot/source/feeds/packages/lang/python/python-pycparser'
time: package/feeds/packages/python-pycparser/python/compile#1.67#0.09#2.26
package/Makefile:107: recipe for target 'package/feeds/packages/python-pycparser/compile' failed
make[1]: *** [package/feeds/packages/python-pycparser/compile] Error 2
make[1]: Leaving directory '/home/brucet/buildbot/source'
/home/brucet/buildbot/source/include/toplevel.mk:216: recipe for target 'package/python-pycparser/compile' failed
make: *** [package/python-pycparser/compile] Error 2

```

jefferyto, 2019-05-24:
If you look at the [x86_64 faillogs](https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/), you will see that python-pycparser is not there. That means the buildbots were able to compile the package successfully. If you can find a faillog for it in any other architecture, I would be interested to know.

You can also find the package available for download for [x86_86](https://downloads.openwrt.org/snapshots/packages/x86_64/packages/). I'm not sure how often a full rebuild is done, from that page you can see the package was last compiled on 27 Apr. If you look at other architectures (e.g. [arm_cortex-a7_neon-vfpv4](https://downloads.openwrt.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/)), you can see that it has been successfully compiled more recently.

> I assume it is a bug in the latest OpenWrt source tree.

I'm sorry, but based on our past email exchanges, I would assume it is your build environment.

You say you are using the x86-generic sdk, but the build log you posted is for a mips target.

I made a VirtualBox Debian VM for you (I'm not going to post the link here). Did you try compiling pycparser there?

If you insist on using your current build environment, at least delete your sdk directory and start over. Compile with full verbosity (`make package/python-pycparser/compile V=s`) and post the full build log (including output for all other packages built).

jefferyto, 2019-05-24:
@commodo Not sure if you have anything else to add.

brucealthompson, 2019-05-25:
I have compiled all existing OpenWrt python packages (except this one) on 2 architectures (x86 and MIPS) as well as 15 new python packages I ported using this build environment. I have also been able to build and run large Python application (Tahoe-LAFS) using this same build environment.

I simply thought I would report this problem formally since it was never addressed in our email thread.

Sounds like you are not interested in examining the root causes of this this issue. It is not a problem for me since I have a workaround for this issue which I described above. I will let someone else reopen this issue when they run into the same problem.

jefferyto, 2019-05-25:
If you refuse to try the suggestions I have provided (and post the information I requested), then there is little else I can do for you.

If you are not willing to examine your own assumptions then there is little anyone can do for you.

jefferyto, 2019-05-25:
BTW pycparser has been in the repo since 2015 (168975c82d15ac8c239bcc0a444c15ae26475cf9). Do you really think no one else has tried to compile it?

brucealthompson, 2019-05-25:
BTW pycparser has been in the repo since 2015 (168975c). Do you really think no one else has tried to compile it?

You are right. I was able to build it using the sources from the 18.04 release with no issue. I think the issue has to do with a more recent change.

 

                Bruce T

brucealthompson, 2019-05-25:
Jeffrey,

To get to the root cause of this issue, I would like to compare my build environment to a build environment in which this package builds. You sent me a virtual machine image of a build environment which you said works via a dropbox link on 4/30. Unfortunately, I did not copy that file to my local machine and the dropbox link is no longer valid.

Can you please make this VM image available again via dropbox or another method?

                Bruce T

commodo, 2019-05-25:
> I'm sorry, but based on our past email exchanges, I would assume it is your build environment.

I also suspect this is the case here.
Please do not try too many things at once.
If it helps, maybe try to setup a new docker image (on your local machine) every time you build something.
As I've found, setting up docker is not too difficult on Ubuntu/Debian distros

brucealthompson, 2019-05-25:
Jeffery / Alexandru,

 

I looked into this issue further.

 

I understand now why my workaround of installing the ply package in the OpenWrt staging directory (staging_dir/hostpkg/lib/python{2.7,3.7}/site-packages) works.

 

The Readme.rst file the pycparser package points out that pycparser depends on the ply library. The Readme.rst file specifically says:

* **pycparser** has no external dependencies. The only non-stdlib library it uses is PLY, which is bundled in ``pycparser/ply``. The current PLY version is

  3.10, retrieved from `<http://www.dabeaz.com/ply/>`_

 

The Readme.rst file also talks about a dependency on parsing tables from the ply library. 

 

In setup.py for pycparser, I saw that by default it tries to build what I believe are the parsing tables from ply. Here is the relevant code from setup.py:

def _run_build_tables(dir):

    from subprocess import check_call

    # This is run inside the install staging directory (that had no .pyc files)

    # We don't want to generate any.

    # https://github.com/eliben/pycparser/pull/135

    check_call([sys.executable, '-B', '_build_tables.py'],

               cwd=os.path.join(dir, 'pycparser'))

 

Notice from the comments that this function is run in the staging directory. When I ran setup.py from a linux host machine using a python debugger, I saw that that “dir” parameter for _run_build_tables was pointing to the python interpreter’s site package directory “…/lib/python2.7/site-packages”. In an OpenWrt cross compile, I believe the “dir” parameter will be set to “…/staging_dir/hostpkg/lib/python{2.7,3.7}/site-packages”. This code ends up compiling the python file “_build_tables.py”. Based on the “dir” parameter, I believe the python interpreter used in the OpenWrt cross compile environment will be looking for packages in the “…/staging_dir/hostpkg/lib/python{2.7,3.7}/site-packages” directory. The python file “_build_tables.py” ends up trying to import the function yacc from ply in the staging directory and fails. So the issue that causes the build of this package to fail is there is no host version of ply installed in the “…/staging_dir/hostpkg/lib/python{2.7,3.7}/site-packages” directory.

 

I have no idea why other people besides myself have not seen this same problem. I am at a complete loss to see how the cross compile build environment determines whether or not a particular OpenWrt host package is installed in the “…/staging_dir/hostpkg/lib/python{2.7,3.7}/site-packages” directory.

 

If I was going to fix this problem, I would simply create a host build rule in the python-ply package Makefile and include a dependency on the python-ply host package in the python-pycparser Makefile. I am not the maintainer of these packages, so I will leave it to your best judgement on how to fix this problem.

 

                Bruce T

brucealthompson, 2019-05-26:
Here is more information on the issue from examining my build environment. I think I may be close to the core of the issue.

It looks like setup.py for pycparser is trying to install a host version of the ply package into the "site-packages" directory before executing “_build_tables.py”. I don't know why, but in my build environment, it is installing the ply package into:
$(STAGING_DIR_HOSTPKG)/$(STAGING_DIR_HOSTPKG)/lib/python{2.7,3.7}/site-packages
instead of:
$(STAGING_DIR_HOSTPKG)/lib/python{2.7,3.7}/site-packages

When I moved the built versions of ply from :
$(STAGING_DIR_HOSTPKG)/$(STAGING_DIR_HOSTPKG)/lib/python{2.7,3.7}/site-packages
to:
$(STAGING_DIR_HOSTPKG)/lib/python{2.7,3.7}/site-packages
the build worked.

I think this line which was executed by  python-pycparser/Makefile is the culprit:
ARCH="x86_64" CC="gcc" CCSHARED="gcc -fPIC" CXX="g++" LD="gcc" LDSHARED="gcc -shared" CFLAGS="-O2 -I/home/brucet/buildbot/source/staging_dir/host/include -I/home/brucet/buildbot/source/staging_dir/hostpkg/include -I/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/host/include" CPPFLAGS="-I/home/brucet/buildbot/source/staging_dir/host/include -I/home/brucet/buildbot/source/staging_dir/hostpkg/include -I/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/host/include -I/home/brucet/buildbot/source/staging_dir/hostpkg/include/python3.7" LDFLAGS="-L/home/brucet/buildbot/source/staging_dir/host/lib -L/home/brucet/buildbot/source/staging_dir/hostpkg/lib -L/home/brucet/buildbot/source/staging_dir/target-mips_mips32_musl/host/lib -lpython3.7 -Wl,-rpath=/home/brucet/buildbot/source/staging_dir/hostpkg/lib" _PYTHON_HOST_PLATFORM=linux2 /home/brucet/buildbot/source/staging_dir/hostpkg/bin/pip3.7 install --root=/home/brucet/buildbot/source/staging_dir/hostpkg --prefix="" "ply==3.10"

From this line, the following options are quite interesting:
--root=/home/brucet/buildbot/source/staging_dir/hostpkg
-rpath=/home/brucet/buildbot/source/staging_dir/hostpkg/lib"

The concatenation of --root and -rpath exactly matches where the ply package ended up getting installed.

Are you still thinking that this issue is due to my build environment??

jefferyto, 2019-05-27:
> Are you still thinking that this issue is due to my build environment??

Yes. `-Wl` passes an option to the linker and `-rpath` tells the linker to "Add a directory to the runtime library search path." (from the manpage for `ld`) This has nothing to do with where the package is installed. `--root` and `--prefix` tell pip where to install the package.

Please do these two things:

1.  At the terminal, type these commands exactly:
    ```
    set > env.txt
    pip list -v > pip_list.txt
    pip3 list -v > pip3_list.txt
    ```
    Paste the contents env.txt, pip_list.txt, and pip3_list.txt here.

2.  At the terminal, type these (and only these) commands exactly:
    ```
    wget https://downloads.openwrt.org/snapshots/targets/ath79/generic/openwrt-sdk-ath79-generic_gcc-7.4.0_musl.Linux-x86_64.tar.xz
    tar Jxf openwrt-sdk-ath79-generic_gcc-7.4.0_musl.Linux-x86_64.tar.xz
    cd openwrt-sdk-ath79-generic_gcc-7.4.0_musl.Linux-x86_64
    ./scripts/feeds update -a
    ./scripts/feeds install -a
    make defconfig
    make package/python-pycparser/compile V=s | tee pycparser.log
    ```
    Paste the content of pycparser.log here.

commodo, 2019-05-27:
I may be an idiot, but I am having trouble following the trail of data.
There is a lot of information here, and I can't identify which is useful.
I would need to dig into things and follow them myself as well, but since I don't get a build problem, I don't know what I would find/see.

I know this sounds not-ok, but since I am not seeing a problem on my end, and since no-one else is not seeing the problem, I will not allocate time/energy to look into this.

Luckily (if there is an issue, then), it's likely not a life threatening issue to raise the priority on this on my work queue (which has grown recently).

brucealthompson, 2019-05-27:
Jeffrey,

Here's the contents of the first 3 files. I am working on the second half now.
[pip_list.txt](https://github.com/openwrt/packages/files/3224337/pip_list.txt)
[pip3_list.txt](https://github.com/openwrt/packages/files/3224339/pip3_list.txt)
[env.txt](https://github.com/openwrt/packages/files/3224340/env.txt)

brucealthompson, 2019-05-27:
Here is the last file. The build failed in the same way, so I assume there is something wrong with my build environment. You are right again.

One comment here. IMHO, it is extremely difficult to set up a proper cross compile build environment for building OpenWrt python packages. You shared a Virtualbox virtual machine which had a proper build environment for OpenWrt / python already install on 4/30. I think it would be extremely helpful if this virtual machine image was made available publicly. This would allow people who are new to OpenWrt to quickly and efficiently set up a build environment for OpenWrt / python that works out of the box.
[pycparser.log](https://github.com/openwrt/packages/files/3224480/pycparser.log)

jefferyto, 2019-05-28:
I was able to reproduce your build results, by installing ply on my build machine (for my local user, not sure about system-wide):

With ply installed for Python 2:
* python-pycparser (Py2) would compile correctly, but host pip doesn't actually install ply (I assume it uses the build machine copy of ply)
* python3-pycparser fails to compile, host pip installs ply to `$(STAGING_DIR_HOSTPKG)/$(STAGING_DIR_HOSTPKG)/lib/python3.7/site-packages`

With ply installed for Python 3:
* python-pycparser (Py2) fails to compile, host pip installs ply to `$(STAGING_DIR_HOSTPKG)/$(STAGING_DIR_HOSTPKG)/lib/python2.7/site-packages`
* Not sure about python3-pycparser, the build didn't get that far but I assume it would succeed, with host pip not actually installing ply

So it looks like host Python/pip is picking up packages installed for the build machine; obviously this shouldn't happen.

I suspect this is also the reason why you were having trouble before, from your pip list output you have a lot of Python packages installed on your build machine and they may be interfering with the sdk's Python.

commodo, 2019-05-28:
Ok, so now we have something concrete to look into.

Well, this is something that keeps re-occurring with different python packages.
I am referring to `setup.py` logic (in some package) that is written without much consideration for cross-compilation.
Well, this isn't bad; it's just how it is :)

jefferyto, 2019-05-28:
@brucealthompson Can you delete your pip cache folder (should be `~/.cache/pip`) then try the build again?

brucealthompson, 2019-05-28:
@jefferyto I just deleted my pip cache folder ( ~/.cache/pip) and tried the build again. It worked!!!!

I am at a loss as to how the pip cache can cause compile errors in the OpenWrt cross compile environment. Nothing surprises me anymore.

Is there anything I can do to keep this type of thing from happening in the future?

Thanks,
Bruce T

jefferyto, 2019-05-29:
AFAICT pip gets confused when it's cache is shared by too many instances of pip.

I'm working on some changes to better isolate it from the local system.

commodo, 2019-05-29:
is there an option to specify a different cache for pip when doing Python host-build ?

jefferyto, 2019-06-04:
Instead of continuing with a virtual machine, I decided to make this: https://github.com/jefferyto/openwrt-vivarium

commodo, 2019-06-05:
looks neat;
can you publish this on dockerhub ( https://hub.docker.com/ ) ?
pricing-wise it looks ok:
https://hub.docker.com/pricing

i mean for it's 1 free, so if you don't need more than 1, it should be ok

the advantage to using dockerhub  is that it can be integrated with Travis CI as well (if that ever becomes an option again);
or even buildbot via docker CLI

jefferyto, 2019-06-05:
Actually I deliberately made it a local image because it downloads the SDK configured in `docker-compose.yml`.

I suppose if there is a way to get Docker Hub to rebuild the image every day (to get the new snapshot SDK), it could also work.

brucealthompson, 2019-06-05:
Very cool. I will try this on Ubuntu.

 

Thanks greatly,

 

                Bruce T

commit ae80ddc7 (2019-06-03): python,python3: Update host pip[3] install functions

* Add --cache-dir option to set the pip cache to a directory in
$(DL_DIR), instead of pip's default (build user's ~/.cache/pip),
fixes #9066

* Add --disable-pip-version-check option, since the version check only
prints a message saying a new version is available

* Combine host_python_pip_install and host_python_pip_install_host into
Build/Compile/HostPy[3]PipInstall

* Remove --root and --prefix options, since this function is only used
to install packages to host Python's default site-packages directory
(setting these may serve to confus [... cut]