openwrt/packages #9080: bluez-examples: Python 3 transition

jefferyto, 2019-05-27:
Maintainer: none
Environment: N/A

Description:
Currently, the bluez-examples package depends on Python 2. As you may be aware, we are transitioning programs to Python 3 (#8893).

I've taken a look at the scripts, and most of them are hard-coded to use Py2 (`#!/usr/bin/python`) while a few (e.g. example-gatt-client) uses Py3 (`#!/usr/bin/env python3`). The ones that use Py2 appear to have taken Py3 into account, but without testing each one I can't say whether they are fully compatible with Py3 or not.

Looking at other distros, I've only found Debian ([bluez-test-scripts](https://packages.debian.org/buster/bluez-test-scripts), [bluez-test-tools](https://packages.debian.org/buster/bluez-test-tools)) and Ubuntu ([bluez-tests](https://packages.ubuntu.com/disco/bluez-tests)) that ship these scripts, but their packages don't have a Python dependency. [Fedora](https://apps.fedoraproject.org/packages/bluez/), [Arch](https://www.archlinux.org/packages/extra/x86_64/bluez/), and [Alpine](https://pkgs.alpinelinux.org/package/edge/main/x86/bluez) don't appear to ship these scripts.

I'd like to get some input on how to proceed. Since there is no current maintainer, pinging recent committers @cotequeiroz @diizzyy @br101 (other members please feel free to chime in).

Should we:
* Change the package dependency and patch the scripts to use Python 3 (possibly breaking some)?
* Remove the Python dependency (and let the user "figure it out")?
* Do something else?

br101, 2019-05-27:
I would suggest to just **not** package these scripts. Just drop the bluez-examples package... I don't find useful scripts in there...

diizzyy, 2019-05-27:
Yeah, drop examples

neheb, 2019-05-27:
I vote for removal as well.

jefferyto, 2019-05-28:
I guess the tribe has spoken :laughing: : #9094

jefferyto, 2019-05-29:
#9094 merged, closing.

closed 2019-05-29 by jefferyto