openwrt/packages #9132: [openssh] client fails to listen on ipv6 socket at boot time

n8v8R, 2019-06-03:
Maintainer: @tripolar 
Environment: 18.06

/etc/ssh/sshd_config
```
Protocol 2
Port 54478
AddressFamily any
ListenAddress 192.168.112.12
ListenAddress fd30:d64c:1eed:4c3a::12
```
___
Description:

upon system boot `sshd` has the ipv4 socket up but not the ipv6 socket. Once the system is fully booted the ipv6 socket for `sshd` becomes available only after _manually restarting_ `sshd`.

This likely being caused by `sshd` starting prior the governing ipv6 iface is fully up. Supposedly `sshd` should only start after the target iface is fully online.

To my understanding the init system (`procd`) or network manager (`netifd`) are not featuring dependencies like targets in `systemd-networkd` or services in `openrc`. 

What works on my node is to change the `procd` S level from 50 to 80. Thus far did not observe a detrimental impact from that change.

neheb, 2019-07-02:
Sounds legit. Can you submit a pull request?

n8v8R, 2019-07-02:
I do not understand why feature suggestions are not always but frequently being met with an invitation to PR? There are plenty of users with no coding skills but nonetheless deploying OpenWrt and contributing to development with bug discovery/reporting and/or feature suggestion.

I do understand that maintenance is time/effort but do you not reckon that it would have cost less to make a PR than making this suggestion in the first place, if I were into coding? Notwithstanding, adding to it now being asked for PR?

Let me put another way. I thought this being useful for others too. since I can manage the setting with a PR or no PR. And if the maintainer cannot be bothered to change a single character in the code then I am wasting my breadth. Closing.

closed 2019-07-02 by n8v8R

neheb, 2019-07-02:
Something like this requires no coding skills. GitHub itself has an edit button.

Put another way, it's much easier to hit the merge button than it is to spend time fixing some issue.

jow-, 2019-07-02:
The change from 50 to 80 will not fix the underlying problem on slower hardware. It only works by chance on fast boards / wan protocols with short setup time.

n8v8R, 2019-07-02:
that is an inhabitation of init then since it does not provide for dependencies of services being up.

80 is still better than 50. If it is not sufficient raise it to 120. Not sure how WAN figures into starting sshd - PPPoE with DSL sync takes like 35 sec from boot t+0 and sshd on ipv6 is available prior PPPoE is established.

jow-, 2019-07-02:
The maximum start limit value is 99. Still, raising it to 80 or 99 is not a proper fix, it just makes the race condition marginally less likely to occur.