openwrt/packages #9133: [lm-sensors] hwmon_device_register() is deprecated

n8v8R, 2019-06-03:
Maintainer: @jow- 
Environment: 18.06

{"kernel":"4.14.122","hostname":"to","system":"ARMv7 Processor rev 1 (v7l)","model":"Turris Omnia","board_name":"armada-385-turris-omnia","release":{"distribution":"TurrisOS","version":"4.0-beta2","revision":"b7132d7","target":"mvebu\/cortexa9","description":"TurrisOS 4.0-beta2 b7132d7"}}
___
Description: kernel log 

> (NULL device *): hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().

referenced https://github.com/torvalds/linux/blob/master/drivers/hwmon/hwmon.c#L741

neheb, 2019-06-03:
Next release will fix this.

n8v8R, 2019-06-04:
next release (19.x) of OpenWRT or of the package?

neheb, 2019-06-04:
Never mind. This error comes from the kernel, not lm-sensors. I see that you are using TurrisOS. Take the issue with the TurrisOS people.

n8v8R, 2019-06-04:
The kernel is provisioned by OpenWRT feeds as is `lm-sensors`. So I report to them and they report it to upstream, just what I did.

If it gets fixed upstream it will reach downstream, or downstream no supposed to fix it?

closed 2019-06-04 by n8v8R

Cynerd, 2019-06-05:
First of all, @neheb and other OpenWRT developers, I would appreciated if you would stop ignoring Turris OS users. Turris OS 4.x+ is based on OpenWRT with just few patches where most of those patches are just backports. Otherwise it is just plain OpenWRT with specific configuration. Otherwise does that mean that anyone who builds OpenWRT on its own is on its own? No support from OpenWRT community and developers? I don't think so. Nobody asks you to support our packages nor out patches but it is madness to ask us on the other hand to support whole OpenWRT. Where is difference between reporting issue by one of Turris developers and one of Turris users? There is none I think. This can be cooperation for benefit of both communities but this kind of approach just slams the doors.

On the other hand I have to reiterate what I wrote in our bug tracker: https://gitlab.labs.nic.cz/turris/turris-build/issues/25#note_107454
@n8v8R there is nothing to fix. This issue is with neither TOS nor with OpenWRT. I grepped all patches in mvebu architecture (both ours and OpenWRT ones) and `hwmon_device_register` is not used not in a single one of them. If it should be reported somewhere then it should be reported to Linux kernel developers but looking in to the latest Linux kernel then this function is still available. I don't think that it is an issue when upstream have not yet phased out. As I already once wrote: **Discussed message is warning for developer not an issue**. So the real reason why this issue should be closed (not because it is TOS) is because there is no issue there.

BKPepe, 2019-06-05:
@neheb as it was said this issue is not Turris OS specific. I just looked at OpenWrt forum and found the issue, which is reported by OP in outputs from different people:

https://forum.openwrt.org/t/iphone-tethering-v18-06-01/24296/3
https://forum.openwrt.org/t/wrt32x-5ghz-wifi-not-associated/33937/6
https://forum.openwrt.org/t/boot-from-an-openwrt-hard-drive-wrt1200ac/33634
https://bugs.openwrt.org/index.php?do=details&task_id=1837

But nobody from them noticed it as an issue so far.

neheb, 2019-06-05:
Right. I've gotten this warning on my Omnia before as well. I never looked into whether it was caused by a standard or non standard config.

While I currently use TOS4, I never looked up how similar or different it is from normal OpenWrt. Previously when I used version 3 I noticed the code was based on OpenWrt DD with a bunch of LEDE patches.

In any case, it seems the issue was reported to the Turris issue tracker prior to being posted here. This was not mentioned.