openwrt/packages #9143: gst1-plugins-bad : mpegtsdemux fails to install

bhargavbhoot, 2019-06-04:
Maintainer: W. Michael Petullo <mike@flyn.org>
                   Ted Hess <thess@kitschensync.net>

Environment: (put here arch, model, OpenWrt version)

Description: enabling the mpegtsdemux fails with the following message : 

Package gst1-mod-mpegtsdemux is missing dependencies for the following libraries:
libgstcodecparsers-1.0.so.0
libgstmpegts-1.0.so.0

I have checked in the build_dir mpegtsdemux, mpegts and codecparsers are getting compiled. but anyhow the plugin mpegtsdemux is not getting installed.

```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2019-07-02:
Is this with latest master or 18.06?

neheb, 2020-02-07:
Closing due to timeout.

closed 2020-02-07 by neheb