openwrt/packages #9148: luci : DHCP leases do not show in LuCI overview when using odhcpd

knacky, 2019-06-04:
On a fresh clone from OpenWrt trunk yesterday, I compiled for x86_64. Deselect dnsmasq, select odhcpd full, select LuCI.

LuCI status overview page at Active DHCP Leases section hangs forever at “Collecting data...”. Yet /tmp/hosts/odhcpd file is populated with devices names and ipv4/ipv6 lease info like normal.

This only started happening in builds from the last 24 hours.

OpenWrt Flyspray user tapper82 commented:

tapper82 commented on 04.06.2019 03:45

Log shows:

Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: Failed to execute call dispatcher target for entry '/admin/dhcplease_status'.
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: The called action terminated with an exception:
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: /usr/lib/lua/luci/tools/status.lua:95: attempt to call global 'duid_to_mac' (a nil value)
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: stack traceback:
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: /usr/lib/lua/luci/tools/status.lua:95: in function </usr/lib/lua/luci/tools/status.lua:9>
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: (tail call): ?
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: /usr/lib/lua/luci/controller/admin/index.lua:140: in function </usr/lib/lua/luci/controller/admin/index.lua:133>
Tue Jun 4 09:44:16 2019 daemon.err uhttpd[4119]: (tail call): ?

knacky, 2019-06-04:
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: Failed to execute call dispatcher target for entry '/admin/dhcplease_status'.
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: The called action terminated with an exception:
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: /usr/lib/lua/luci/tools/status.lua:95: attempt to call global 'duid_to_mac' (a nil value)
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: stack traceback:
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: 	/usr/lib/lua/luci/tools/status.lua:95: in function </usr/lib/lua/luci/tools/status.lua:9>
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: 	(tail call): ?
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: 	/usr/lib/lua/luci/controller/admin/index.lua:140: in function </usr/lib/lua/luci/controller/admin/index.lua:133>
Tue Jun  4 09:15:14 2019 daemon.err uhttpd[3121]: 	(tail call): ?

dibdot, 2019-06-04:
fixed in master right now - thanks for reporting this issue.

closed 2019-06-04 by dibdot