openwrt/packages #9158: dnscrypt proxy 1.9.5

jglooije, 2019-06-05:
Latest libsodium update to 1.0.18 causes error in excecution:
libsodium change log related to error
- randombytes_salsa20 has been renamed to randombytes_internal

porroso, 2019-06-21:
same error here:
daemon.err uhttpd[3008]: Error relocating /usr/sbin/dnscrypt-proxy: randombytes_salsa20_implementation: symbol not found

compilied rev Openwrt-ar71xx-r10263_gcc8.3.0_kern_4.14.127

porroso, 2019-06-22:
thanks @neheb !! problem solved after git pull and manually add your patch net/dnscrypt-proxy/patches/010-internal.patch

neheb, 2019-06-22:
Cool. Now just waiting for maintainer to ACK. After that I will backport to 19.07 branch as it's broken there.

neheb, 2019-07-02:
Fixed and backported to 19.07.

closed 2019-07-02 by neheb