openwrt/packages #9194: avrdude: bad %ud formattiong in linuxgpio

jh-cg, 2019-06-09:
Maintainer: @thess 
Environment: ar71xx, Mikrotik mAP lite, OpenWRT 18.06.2

Description:
The driver `linuxgpio` in file `linuxgpio.c` uses `%ud` to format numbers when building strings (which are then used as filenames and to interface with `/sys`). This format string is not understood by openwrt libc, the number becomes `<number>d`, and this leads to various errors.

Running find and replace `%ud` with `%u` on `linuxgpio.c` and rebuilding the package fixed the problem.

I'm not sure whether this should be considered as an openwrt or upstream bug (as %ud is AFAIK rather uncommon).

neheb, 2019-06-09:
Upstream issue yes.

jh-cg, 2019-06-09:
Oh, it seems to be already reported (3 years ago). https://savannah.nongnu.org/bugs/?49450

neheb, 2019-06-09:
In the meantime, I recommend submitting a pull request that fixes this.

neheb, 2019-07-02:
Added a pull request. Test if you can.

neheb, 2020-01-24:
I fixed this. Closing.

closed 2020-01-24 by neheb