openwrt/packages #9200: [qrencode] cannot generate svg image

n8v8R, 2019-06-09:
Maintainer: @jp-bennett 
Environment: 18.06 && Master

Description: generating a svg token image with

```
qrencode "foo://$(ubus call network.interface.lan status | sed -n 's|.*"address": "\([0-9.]*\)"
.*|\1|p')/luci-static/my_secret_url.token?scheme=http&hostname=$HOSTNAME&board_name=bar" -o /www/luci-static/my_secret_url.svg
```

fails with

> PNG output is disabled at compile time. No output generated.
___
Apparently the package is compiled with `--without-png`. Not sure what the merit of this flag is but it would be appreciated to remove it in order to be able to generate svg token images

dibdot, 2019-06-09:
You have to define the output type (which default is png) with `--type=SVG`

n8v8R, 2019-06-09:
fabulous

closed 2019-06-09 by n8v8R

firewire10000, 2021-07-03:
How do I compile qrencode with PNG support if I'm compiling my own OpenWrt firmware?