openwrt/packages #9201: adblock: unbound restarting on adblock reload even though Flush DNS Cache is off

mpeleshenko, 2019-06-09:
Whenever I run adblock reload, I see that unbound is restarted even though I turned off Flush DNS Cache. It seems the below line should check for `[ ${adb_dnsflush} -eq 1]`.

https://github.com/openwrt/packages/blob/1c4b6fef5f57364d65fda1c1e072cc960e9d3dfb/net/adblock/files/adblock.sh#L425

mpeleshenko, 2019-06-09:
@dibdot

dibdot, 2019-06-09:
Did you install unbound-control as well? Whenever your blocklists changes during an adblock run, a reload via unbound-control takes place (which preserves the cache) or - if unbound-control is not available - a restart happens.

closed 2019-06-09 by dibdot

brianjmurrell, 2023-01-06:
Can this be reopened please?  I also do not have  enabled and I **do** have `unbound-control` installed but am also seeing unbound restarted when Adblock is reloaded/restarted:

```
Jan  5 22:11:18 example.com adblock-4.1.5[6109]: adblock instance started ::: action: reload, priority: 0, pid: 6109
Jan  5 22:11:46 example.com unbound: [7437:0] notice: init module 0: iterator
Jan  5 22:11:46 example.com unbound: [7437:0] info: start of service (unbound 1.17.0).
Jan  5 22:12:40 example.com unbound: [7437:0] info: service stopped (unbound 1.17.0).
Jan  5 22:12:40 example.com unbound: [7437:0] info: server stats for thread 0: 35 queries, 9 answers from cache, 26 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Jan  5 22:12:40 example.com unbound: [7437:0] info: server stats for thread 0: requestlist max 6 avg 2.23077 exceeded 0 jostled 0
Jan  5 22:12:40 example.com unbound: [7437:0] info: average recursion processing time 0.100639 sec
Jan  5 22:12:40 example.com unbound: [7437:0] info: histogram of recursion processing times
Jan  5 22:12:40 example.com unbound: [7437:0] info: [25%]=0.0444709 median[50%]=0.0786432 [75%]=0.121242
Jan  5 22:12:40 example.com unbound: [7437:0] info: lower(secs) upper(secs) recursions
Jan  5 22:12:40 example.com unbound: [7437:0] info:    0.016384    0.032768 4
Jan  5 22:12:40 example.com unbound: [7437:0] info:    0.032768    0.065536 7
Jan  5 22:12:40 example.com unbound: [7437:0] info:    0.065536    0.131072 10
Jan  5 22:12:40 example.com unbound: [7437:0] info:    0.131072    0.262144 3
Jan  5 22:12:40 example.com unbound: [7437:0] info:    0.262144    0.524288 2
Jan  5 22:12:44 example.com unbound: [8528:0] notice: init module 0: iterator
Jan  5 22:12:44 example.com unbound: [8528:0] info: start of service (unbound 1.17.0).
Jan  5 22:12:45 example.com adblock-4.1.5[7965]: adblock instance started ::: action: restart, priority: 0, pid: 7965
Jan  5 22:12:48 example.com unbound: [8528:0] info: service stopped (unbound 1.17.0).
Jan  5 22:12:48 example.com unbound: [8528:0] info: server stats for thread 0: 4 queries, 0 answers from cache, 4 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Jan  5 22:12:48 example.com unbound: [8528:0] info: server stats for thread 0: requestlist max 2 avg 0.75 exceeded 0 jostled 0
Jan  5 22:12:48 example.com unbound: [8528:0] info: average recursion processing time 0.229603 sec
Jan  5 22:12:48 example.com unbound: [8528:0] info: histogram of recursion processing times
Jan  5 22:12:48 example.com unbound: [8528:0] info: [25%]=0.098304 median[50%]=0.131072 [75%]=0.393216
Jan  5 22:12:48 example.com unbound: [8528:0] info: lower(secs) upper(secs) recursions
Jan  5 22:12:48 example.com unbound: [8528:0] info:    0.065536    0.131072 2
Jan  5 22:12:48 example.com unbound: [8528:0] info:    0.262144    0.524288 2
Jan  5 22:12:52 example.com unbound: [9472:0] notice: init module 0: iterator
Jan  5 22:12:52 example.com unbound: [9472:0] info: start of service (unbound 1.17.0).
Jan  5 22:14:15 example.com unbound: [9472:0] info: service stopped (unbound 1.17.0).
Jan  5 22:14:15 example.com unbound: [9472:0] info: server stats for thread 0: 225 queries, 23 answers from cache, 202 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Jan  5 22:14:15 example.com unbound: [9472:0] info: server stats for thread 0: requestlist max 59 avg 11.5693 exceeded 0 jostled 0
Jan  5 22:14:15 example.com unbound: [9472:0] info: average recursion processing time 0.563685 sec
Jan  5 22:14:15 example.com unbound: [9472:0] info: histogram of recursion processing times
Jan  5 22:14:15 example.com unbound: [9472:0] info: [25%]=0.18787 median[50%]=0.401582 [75%]=0.847093
Jan  5 22:14:15 example.com unbound: [9472:0] info: lower(secs) upper(secs) recursions
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.000000    0.000001 2
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.016384    0.032768 3
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.032768    0.065536 3
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.065536    0.131072 23
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.131072    0.262144 45
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.262144    0.524288 47
Jan  5 22:14:15 example.com unbound: [9472:0] info:    0.524288    1.000000 42
Jan  5 22:14:15 example.com unbound: [9472:0] info:    1.000000    2.000000 34
Jan  5 22:14:15 example.com unbound: [9472:0] info:    2.000000    4.000000 3
Jan  5 22:14:22 example.com unbound: [11826:0] notice: init module 0: iterator
Jan  5 22:14:23 example.com unbound: [11826:0] info: start of service (unbound 1.17.0).
Jan  5 22:14:27 example.com adblock-4.1.5[7965]: blocklist with overall 51232 blocked domains loaded successfully (TP-Link Archer C7 v2, OpenWrt 22.03.2 r19803-9a599fee93)
```

As an aside, it took that many adblock restart/reload attempts to get it to finally block domains.  But I suppose that's another ticket.  The above is about the unbound restarts due to the adblock restart/reloads even though _Flush DNS Cache_ is disabled.

brianjmurrell, 2025-04-12:
Second request for reopening this issue per above.

dibdot, 2025-04-12:
Why? What did you expect? If you use the adblock package with Unbound, it will run the unbound-control dump_cache and load_cache commands when restarting the instance.

brianjmurrell, 2025-04-13:
My previous request was:

> but am also seeing unbound restarted when Adblock is reloaded/restarted

But perhaps that is just when adblock is reloaded and not also when restarted.

Is it expected that adblock needs to restart unbound when it is restarted?  Why if so?  Why isn't unbound reload sufficient for all adblock state changes (i.e. start, stop, restart, etc.)?

dibdot, 2025-04-13:
> Why isn't unbound reload sufficient

There is no difference between reload and restart (at least for unbound), e.g. the reload output ...
```
Sun Apr 13 10:16:26 2025 daemon.info unbound: [15064:0] info: service stopped (unbound 1.22.0).
[...]
Sun Apr 13 10:16:40 2025 daemon.info unbound: [15498:0] info: start of service (unbound 1.22.0).
```
As wrote above, unbound-control is doing the magic and preserve the cache between restarts.

brianjmurrell, 2025-04-13:
Ahhh.  Fair enough.  I guess I was just hoping that unbound reload would actually be a reload and would be less copious about what it logs (stats and whatnot).

Nice feature that adblock dumps the cache before the reload/restart and loads it afterwards BTW.  I have thought about doing that with BIND (in a different context than OpenWRT) in the past.  Never got around to actually implementing it though.