openwrt/packages #9219: "./etc/init.d/dnsmasq: line 25: dnsmasq: command not found"

mistepien, 2019-06-11:
During build image with imagebuilder I got this:

>   Finalizing root filesystem...
/root/openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/dnsmasq: line 25: dnsmasq: command not found
Enabling avahi-daemon
Enabling boot
Enabling bootcount
Enabling cron
Enabling ddns
./etc/init.d/dnsmasq: line 25: dnsmasq: command not found
Enabling dnsmasq

There are how to I prepare image:
`make image PROFILE=ubnt-erx PACKAGES="luci-ssl -wpad-mini -hostapd-common -hostapd -dnsmasq-full -dnsmasq-dhcpv6 haveged avahi-nodbus-daemon luci-proto-ppp  screen pv htop  avahi-daemon-service-ssh tcpdump ip-full iperf iperf3 luci-app-watchcat irqbalance ipset luci-app-ddns nmap-ssl openssh-sftp-server ca-certificates   -odhcp6c -odhcpd -odhcpd-ipv6only  curl ethtool snmpd snmp-utils drill luci-app-unbound unbound-control  openssh-server ntp-utils ntpd  dnsmasq" `

jow-, 2019-06-11:
@yousong - this is due to https://github.com/openwrt/openwrt/commit/0299a4b73e729504dfdc5c4563535db082bdf941 - the `$dnsmasq_features` checking needs to happen in one of the procedures not directly in the toplevel script context.

closed 2019-06-11 by jow-

yousong, 2019-06-11:
Thanks for the report.

@mistepien , updated dnsmasq build will be available after buildbots pick up the change.