openwrt/packages #9229: Bash 5.0.2 build fails on x86_64

gilfrade, 2019-06-13:
Maintainer: @diizzyy 
Environment: 19.07

Description: Fails to build as is. Previous version compiles ok.

```
./lib/sh/libsh.a(tmpfile.o): In function `sh_mktmpname':
tmpfile.c:(.text+0x142): warning: the use of `mktemp' is dangerous, better use `mkstemp' or `mkdtemp'
y.tab.o: In function `shell_getc':
y.tab.c:(.text+0x20a1): undefined reference to `history_quoting_state'
y.tab.c:(.text+0x20b1): undefined reference to `history_quoting_state'
y.tab.c:(.text+0x20bd): undefined reference to `history_quoting_state'
y.tab.c:(.text+0x20de): undefined reference to `history_quoting_state'
bashhist.o: In function `bash_history_inhibit_expansion':
bashhist.c:(.text+0xcf): undefined reference to `history_quoting_state'
bashhist.o: In function `bash_delete_history_range':
bashhist.c:(.text+0x348): undefined reference to `remove_history_range'
bashline.o: In function `bash_execute_unix_command':
bashline.c:(.text+0x2be6): undefined reference to `rl_function_of_keyseq_len'
collect2: error: ld returned 1 exit status
make[3]: *** [bash] Error 1
```

gilfrade, 2019-06-13:
Solved it by cherry picking commit `readline: Update to 8.0 and various fixes` (166b335e6eeabcbcd0a3f817e210a5166dcdb202) to branch 19.07 of Openwrt.  
Before i tried to disable readline in bash makefile but it still failed on history functions.

diizzyy, 2019-06-13:
Hi, 
I'm not sure I follow you as it's already in 19.07? https://github.com/openwrt/openwrt/tree/openwrt-19.07/package/libs/readline

diizzyy, 2019-06-22:
Closing due to submitter timeout

closed 2019-06-22 by diizzyy