openwrt/packages #9235: simple-adblock: fatal error when restarting dnsmasq if an input list contains non-ascii characters

SebastianApel, 2019-06-15:
Maintainer: @stangri 
Environment:  OpenWrt SNAPSHOT r10206-35a70d6 / LuCI Master (f138fc93) // AVM FRITZ!Box 3370 Rev. 2 (Micron NAND)

Description:
Dear Stan,
first: let me say "thank you" for your work on the Openwrt simple-adblock package.

I have had a problem with the package: one of the files (https://someonewhocares.org/hosts/hosts) contains a non-ascii domain name ("secrects.Google.com", but the "capital G" is not an ascii G but another character set).

Downstream, this prevents dnsmasq from starting with an error message:

Sat Jun 15 21:31:10 2019 daemon.crit dnsmasq[5952]: error at line 36247 of /tmp/dnsmasq.d/simple-adblock
Sat Jun 15 21:31:10 2019 daemon.crit dnsmasq[5952]: FAILED to start up

Quick-and-dirty fix: 
I modified the /etc/init.d/simple-adblock script with a quick and dirty hack that filters non-ascii characters (pls find it attached). A better way might be to drop the line completely - but I am not sure how to do that with OpenWRT onboard utils.

[simple-adblock.txt](https://github.com/openwrt/packages/files/3293508/simple-adblock.txt)

Kind regards,
Sebastian

stangri, 2019-06-15:
I've been meaning to submit the pr for the update to the OpenWrt repo, in
the mean time you can install the update from my own repo,
https://stangri.github.io/openwrt-repo/

dibdot, 2019-07-05:
Fixed in master with #9395

closed 2019-07-05 by dibdot