openwrt/packages #9254: libupm worries

neheb, 2019-06-18:
Maintainer: @nxhack 

From IRC:

```
05:35 < jow> wtf is libupm
05:35 < jow> it appears to kill our phase2 builders
05:35 < jow> apparently it takes longer than an our to build it
05:35 < jow> *hour
05:36 < blogic> me hides
05:36 < jow> ah, some iot crapfest
05:36 < blogic> its that intel IoT crap
05:36 < ldir> fest
05:36 < blogic> set it to source-only
05:36 < blogic> ldir: laughs on me
05:37 < jow> can't easily do that
05:37 < ldir> instant karma is quick these days :-)
05:37 < blogic> jow: i'll set it and stuff that depends on it to source-only
05:37 < blogic> jow: how not ?
05:37 < jow> iirc there's no "source only" flag for packages
05:37 < blogic> but an hour is long, used to take 10 minutes
05:37 < jow> just @BROKEN
05:37 < blogic> then break it
05:38 < blogic> i ported it for the linkit smart thingy
05:41 < jow> I think its unrelated
05:41 < jow> libupm itself was touched months ago
05:42 < jow> but right now its spending excessive amounts of time compiling various lsm9ds0PYTHON_wrap.cxx.o files
05:43 < blogic> maybe some pip carnage
05:43 < blogic> like its trying to access some $could$hipster$crap that is offline
05:43 < blogic> i vote for BROKEN
05:44 < stintel> what about implementing source only for packages?
05:48 < jow> added an "scripts/feeds uninstall libupm" for now to stabilize things
06:10 < jow> ynezz: source-only applied
```

neheb, 2019-06-18:
A quick look shows no PKG_BUILD_PARALLEL is set. There's also a newer version.

nxhack, 2019-06-18:
OK I will do that.

nxhack, 2019-06-20:
It seems that it has already been marked BROKEN.
There are many cases where OpenWrt is used as IoT server....

diizzyy, 2019-06-21:
This is sadly a rather non encouraging attitude by upstream especially since it was added by one of the devs and then more or less abandoned.

diizzyy, 2019-06-21:
...and I don't see a need to keep this open.

closed 2019-06-21 by diizzyy