openwrt/packages #9255: seafile-server failing on ARC

neheb, 2019-06-18:
Maintainer: @commodo 

https://downloads.openwrt.org/snapshots/faillogs/arc_arc700/packages/seafile-server/compile.txt

One way I've found to solve this issue is with PKG_SSP:=0. I can't test this as glib2 doesn't build on my setup for some reason.

commodo, 2019-06-19:
FWIW, there is some insight in this commit https://github.com/openwrt/openwrt/commit/241e6dd3e92c4f215b8ac75379a4b5aeaeb92171

neheb, 2019-06-19:
Interesting...

My solution to this problem is here: https://github.com/openwrt/packages/blob/master/utils/powertop/Makefile#L43

I reverted and added -lssp to LDFLAGS and that works as well. I wonder if this also fails with glibc...

commodo, 2019-06-20:
Adding the patch in `include/hardening.mk` also works.
After looking around the current packages, I am not completely sure about patching this in core.
There are only 2-3 packages [in the OpenWrt project] that may benefit from this [i.e. patching this in core]; maybe there may be a few others, but I am not sure.

I'll just patch seafile-server only then.

closed 2019-06-20 by yousong

neheb, 2019-06-20:
I backported this to 19.07. Hope you don't mind.

commodo, 2019-06-20:
No worries from me.

Thanks :)
Alex