openwrt/packages #9258: rp-pppoe-server : Startup and configuration items

Chaskel, 2019-06-18:
Maintainer: @cshoredaniel 
Environment: arm_arm1176jzf-s_vfp / brcm2708 / bcm2708 (Raspberry Pi B+) w/ OpenWrt 18.06.2

Description:
To allow pppoe-server startup, and also be configurable in OpenWrt GUI, I made the following changes. Please note that I am not sure if this is the best way to fix, but thought providing the following would be helpful:

**/etc/init.d/pppoe-server:**
In the following line I remove "device" just before "ac_name".
**OLD:** local enabled interface device ac_name service_names service_name maxsessionsperpeer localip firstremoteip maxsessions optionsfiles randomsession unit offset timeout mss sync OPTIONS
**NEW:** local enabled interface ac_name service_names service_name maxsessionsperpeer localip firstremoteip maxsessions optionsfiles randomsession unit offset timeout mss sync OPTIONS

I combined the next 2 lines to be one and removed reference to "device" just before the "$interface" reference.
**OLD:** . /lib/functions/network.sh
     network_get_physdev device $interface || return 1
**NEW:** . /lib/functions/network.sh network_get_physdev $interface || return 1

In the next line I replaced "$device" with "$interface".
**OLD:** append OPTIONS "-I $device"
**NEW:** append OPTIONS "-I $interface"

**/etc/config/pppoe:**
I had to un-comment all the lines so they could be displayed and be managed via the OpenWrt GUI.
Note: Even though I un-commented the pppoe_relay portion of the file, those do not show up however I do not believe I need to use those for just running the PPPoE server (but could be mistaken).

feckert, 2019-07-21:
Please provide a source change what you have done. So we can see changes what you have done

tinyhare, 2022-05-21:
I can't start PPPoE server，and find the problem：
[image]
[image]

in  /etc/init.d/pppoe-server
```
pppoe_instance(){

    ...
    # the $interface  actually is a device name, so this always return 1 !!!!!!!!!!!!!!!!!
    network_get_physdev device $interface || return 1

    ...
}

```

so I configure the` /etc/config/pppoe `file, ` option interface` to real interface name, and it worked for me.

feckert, 2022-05-23:
Have you tried to replace the template https://github.com/openwrt/luci/blob/master/applications/luci-app-rp-pppoe-server/luasrc/model/cbi/rp-pppoe-server.lua#L16 with `cbi/network_ifacelist`?
I think the problem is that you select a device and not a logical interface. The init script is dynamical looking up the physical interface on start. 
If you change this in LuCI, you still have restart the router.

openwrtdiy, 2022-06-20:
@feckert luci-app-rp-pppoe-server
A bit outdated and needs to be upgraded. I'm testing some updates but having some technical issues.
[image]
[image]
[image]