openwrt/packages #9263: libuhttpd-2.2.1: build on master fails (when building ipq806x/r7500v2)

notmyrealhandle, 2019-06-18:
Maintainer: @zhaojh329 

Environment: ipq806x, r7500v2, master

Description:

building master fails with
```
[ 30%] Building C object src/CMakeFiles/uhttpd.dir/utils.c.o
  CC       sys-utils/switch_root.o
/home/n/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/libuhttpd-\
nossl/libuhttpd-2.2.1/src/utils.c: In function 'uh_chunk_send':
/home/n/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/libuhttpd-\
nossl/libuhttpd-2.2.1/src/utils.c:49:24: error: too many arguments for format [\
-Werror=format-extra-args]
     ustream_printf(us, "\r\n", len);
                        ^~~~~~
/home/n/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/libuhttpd-\
nossl/libuhttpd-2.2.1/src/utils.c: In function 'uh_chunk_vprintf':
/home/n/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/libuhttpd-\
nossl/libuhttpd-2.2.1/src/utils.c:79:24: error: too many arguments for format [\
-Werror=format-extra-args]
     ustream_printf(us, "\r\n", len);
                        ^~~~~~
cc1: all warnings being treated as errors
src/CMakeFiles/uhttpd.dir/build.make:101: recipe for target 'src/CMakeFiles/uht\
tpd.dir/utils.c.o' failed
make[6]: *** [src/CMakeFiles/uhttpd.dir/utils.c.o] Error 1
make[6]: Leaving directory '/home/n/openwrt/build_dir/target-arm_cortex-a15+neo\
n-vfpv4_musl_eabi/libuhttpd-nossl/libuhttpd-2.2.1'
CMakeFiles/Makefile2:92: recipe for target 'src/CMakeFiles/uhttpd.dir/all' fail\
ed
make[5]: *** [src/CMakeFiles/uhttpd.dir/all] Error 2
make[5]: Leaving directory '/home/n/openwrt/build_dir/target-arm_cortex-a15+neo\
n-vfpv4_musl_eabi/libuhttpd-nossl/libuhttpd-2.2.1'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/n/openwrt/build_dir/target-arm_cortex-a15+neo\
n-vfpv4_musl_eabi/libuhttpd-nossl/libuhttpd-2.2.1'
Makefile:83: recipe for target '/home/n/openwrt/build_dir/target-arm_cortex-a15\
+neon-vfpv4_musl_eabi/libuhttpd-nossl/libuhttpd-2.2.1/.built' failed
make[3]: *** [/home/n/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_e\
abi/libuhttpd-nossl/libuhttpd-2.2.1/.built] Error 2
make[3]: Leaving directory '/home/n/openwrt/feeds/packages/libs/libuhttpd'
time: package/feeds/packages/libuhttpd/nossl/compile#0.90#0.26#2.88
package/Makefile:107: recipe for target 'package/feeds/packages/libuhttpd/compi\
le' failed
make[2]: *** [package/feeds/packages/libuhttpd/compile] Error 2
make[2]: *** Waiting for unfinished jobs....
```

my last successful build was 6-7 days ago and it looks like there were related changes to package uhttpd a few days ago.

zhaojh329, 2019-06-19:
Fixed. #9266

notmyrealhandle, 2019-06-19:
thankyou

closed 2019-06-19 by notmyrealhandle