openwrt/packages #9270: python-requests

pawisoon, 2019-06-19:
Is it possible to include python-requests in openwrt packages?
Cheers,
Pawel

jefferyto, 2019-06-19:
It might help to [check](https://github.com/openwrt/packages/blob/master/lang/python/python-requests/Makefile) the repo before asking.

pawisoon, 2019-06-19:
> It might help to [check](https://github.com/openwrt/packages/blob/master/lang/python/python-requests/Makefile) the repo before asking.

Hey, 
I saw it but forgot to mention Im trying to build it under 15.05. I tried using current Makefile with my changes but it failed to compile: 
[image]

jefferyto, 2019-06-21:
15.05 is 4 years old and isn't supported anymore - I would advise you to upgrade.

BKPepe, 2019-06-23:
@jefferyto said everything. Closing.

closed 2019-06-23 by BKPepe