openwrt/packages #9274: openvswitch error with kernel 4.19

neheb, 2019-06-19:
Maintainer: @yousong 
Environment: kernel 4.19

https://downloads.openwrt.org/snapshots/faillogs/powerpc_8540/packages/openvswitch/compile.txt

yousong, 2019-06-20:
@neheb  Thanks for the report.

closed 2019-06-20 by yousong

Leo-PL, 2019-07-01:
@yousong I'm getting a very similar failure on ath79 since openvswitch was bumped. Maybe I need to update my .config somehow?
```
make[3]: Entering directory '/mnt/work/builds/openwrt/ath79/feeds/packages/net/openvswitch'
Makefile:331: XXX: openvswitch: intree kmods selected but not supported
mkdir -p /mnt/work/builds/openwrt/ath79/staging_dir/target-mips_24kc_musl/root-ath79/stamp
SHELL= flock /mnt/work/builds/openwrt/ath79/tmp/.root-copy.flock -c 'cp -fpR /mnt/work/builds/openwrt/ath79/build_dir/target-mips_24kc_musl/linux-ath79_generic/openvswitch-2.11.1/.pkgdir/kmod-openvswitch/. /mnt/work/builds/openwrt/ath79/staging_dir/target-mips_24kc_musl/root-ath79/'
touch /mnt/work/builds/openwrt/ath79/staging_dir/target-mips_24kc_musl/root-ath79/stamp/.kmod-openvswitch_installed
mkdir -p /mnt/work/builds/openwrt/ath79/bin/targets/ath79/generic/packages /mnt/work/builds/openwrt/ath79/build_dir/target-mips_24kc_musl/linux-ath79_generic/openvswitch-2.11.1/ipkg-mips_24kc/kmod-openvswitch/CONTROL /mnt/work/builds/openwrt/ath79/staging_dir/target-mips_24kc_musl/pkginfo
mkdir -p /mnt/work/builds/openwrt/ath79/build_dir/target-mips_24kc_musl/linux-ath79_generic/openvswitch-2.11.1/ipkg-mips_24kc/kmod-openvswitch/etc/modules.d; ( echo "/mnt/work/builds/openwrt/ath79/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-4.19.55/net/openvswitch/openvswitch"; ) > /mnt/work/builds/openwrt/ath79/build_dir/target-mips_24kc_musl/linux-ath79_generic/openvswitch-2.11.1/ipkg-mips_24kc/kmod-openvswitch/etc/modules.d/openvswitch; 
find /mnt/work/builds/openwrt/ath79/build_dir/target-mips_24kc_musl/linux-ath79_generic/openvswitch-2.11.1/ipkg-mips_24kc/kmod-openvswitch -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package kmod-openvswitch is missing dependencies for the following libraries:
nf_conncount.ko
Makefile:338: recipe for target '/mnt/work/builds/openwrt/ath79/bin/targets/ath79/generic/packages/kmod-openvswitch_4.19.55+2.11.1-1_mips_24kc.ipk' failed
make[3]: *** [/mnt/work/builds/openwrt/ath79/bin/targets/ath79/generic/packages/kmod-openvswitch_4.19.55+2.11.1-1_mips_24kc.ipk] Error 1
make[3]: Leaving directory '/mnt/work/builds/openwrt/ath79/feeds/packages/net/openvswitch'
time: package/feeds/packages/openvswitch/compile#0.15#0.06#0.41
package/Makefile:107: recipe for target 'package/feeds/packages/openvswitch/compile' failed
make[2]: *** [package/feeds/packages/openvswitch/compile] Error 2
make[2]: Leaving directory '/mnt/work/builds/openwrt/ath79'
package/Makefile:103: recipe for target '/mnt/work/builds/openwrt/ath79/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/mnt/work/builds/openwrt/ath79/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/work/builds/openwrt/ath79'
/mnt/work/builds/openwrt/ath79/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

yousong, 2019-07-01:
@Leo-PL , thanks for letting me know.  Please give the following patch a try.  Thank you.

```
diff --git a/net/openvswitch/Makefile b/net/openvswitch/Makefile
index 7e7b25cfb..bd23b33d7 100644
--- a/net/openvswitch/Makefile
+++ b/net/openvswitch/Makefile
@@ -75,6 +75,7 @@ ovs_kmod_openvswitch_depends:=\
 	  +kmod-nf-conntrack \
 	  +IPV6:kmod-nf-conntrack6 \
 	  +(!LINUX_4_9&&!LINUX_4.14):kmod-nsh \
+	  +(!LINUX_4_9&&!LINUX_4.14):kmod-ipt-conntrack-extra \
 
 ovs_kmod_openvswitch_files:=$(ovs_kmod_upstream_dir)/openvswitch.ko
 $(eval $(call OvsKmodPackageTemplate,openvswitch))
```

Leo-PL, 2019-07-01:
It worked, now #9273 hit me ;)
Thanks!