openwrt/packages #9275: python3-netifaces: build error caused by linking with -L/usr/lib in configure script

cotequeiroz, 2019-06-19:
Maintainer: @BKPepe 
Environment: mvebu, arm_cortex-a9+vfpu3, current master

Description:
I couldn't find the bug in the bots, but it fails here consistently.  The failure output is this:
```
if [ "" == "HOST" ]; then export PYTHONPATH="/home/equeiroz/src/openwrt/staging_dir/hostpkg/lib/python3.7:/home/equeiroz/src/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages"; export PYTHONDONTWRITEBYTECODE=0; else export PYTHONPATH="/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib/python3.7:/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi//usr/lib/python3.7/site-packages:/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9/ipkg-install//usr/lib/python3.7/site-packages"; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi"; export _python_prefix="/usr"; export _python_exec_prefix="/usr"; fi; export PYTHONOPTIMIZE="";   cd /home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9/; CC="arm-openwrt-linux-muslgnueabi-gcc" CCSHARED="arm-openwrt-linux-muslgnueabi-gcc -fpic" CXX="arm-openwrt-linux-muslgnueabi-g++" LD="arm-openwrt-linux-muslgnueabi-gcc" LDSHARED="arm-openwrt-linux-muslgnueabi-gcc -shared" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/include -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include/fortify -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/python3.7" LDFLAGS="-L/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/lib -L/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/lib -znow -zrelro -lpython3.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python3.7"      /home/equeiroz/src/openwrt/staging_dir/hostpkg/bin/python3.7   ./setup.py   install --prefix="/usr" --root="/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9/ipkg-install" --single-version-externally-managed ;
running install
running build
running build_ext
checking for getifaddrs...not found.
checking for getnameinfo...not found.
checking for socket IOCTLs...not found.
checking for optional header files...netpacket/packet.h linux/atm.h linux/llc.h linux/tipc.h linux/dn.h.
checking whether struct sockaddr has a length field...no.
checking which sockaddr_xxx structs are defined...none!
checking for routing socket support...no.
checking for sysctl(CTL_NET...) support...no.
checking for netlink support...no.
building 'netifaces' extension
arm-openwrt-linux-muslgnueabi-gcc -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.2:Python-3.7.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/include -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include/fortify -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/python3.7 -fPIC -DNETIFACES_VERSION=0.10.9 -DHAVE_NETPACKET_PACKET_H=1 -DHAVE_LINUX_ATM_H=1 -DHAVE_LINUX_LLC_H=1 -DHAVE_LINUX_TIPC_H=1 -DHAVE_LINUX_DN_H=1 -I/home/equeiroz/src/openwrt/staging_dir/hostpkg/include/python3.7 -c netifaces.c -o build/temp.linux2-3.7/netifaces.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
netifaces.c:210:6: error: #error You need to add code for your platform.
 #    error You need to add code for your platform.
      ^~~~~
error: command 'arm-openwrt-linux-muslgnueabi-gcc' failed with exit status 1
make[2]: *** [Makefile:47: /home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9/.built] Error 1
make[2]: Leaving directory '/home/equeiroz/src/openwrt/feeds/packages/lang/python/python3-netifaces'
```
I've checked the build dir, and found out where the problem seems to be:
```
~/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9/build/temp.linux2-3.7 $ cat conftest.out
creating build/temp.linux2-3.7/build
creating build/temp.linux2-3.7/build/temp.linux2-3.7
arm-openwrt-linux-muslgnueabi-gcc -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.2:Python-3.7.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/equeiroz/src/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/include -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include/fortify -I/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include -I/home/equeiroz/src/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/python3.7 -fPIC -I/home/equeiroz/src/openwrt/staging_dir/hostpkg/include/python3.7 -c build/temp.linux2-3.7/conftest-0.c -o build/temp.linux2-3.7/build/temp.linux2-3.7/conftest-0.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
arm-openwrt-linux-muslgnueabi-gcc build/temp.linux2-3.7/build/temp.linux2-3.7/conftest-0.o -L/usr/lib -o build/temp.linux2-3.7/conftest
/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: skipping incompatible /usr/lib/libc.so when searching for -lc
/home/equeiroz/src/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /usr/lib/libc.a: error adding symbols: file format not recognized
collect2: error: ld returned 1 exit status
```
If I link the file manually, omitting `-L/usr/lib`, it succeeds; if I _hack_ `TARGET_CC+= $(TARGET_LDFLAGS)` in the Makefile, the package builds, but I don't believe this is a solution, so I'm not opening a PR.  I'm not familiar with python's package build system, so I could not identify where the flag is being added.  @jefferyto, @commodo, do you have any thoughts regarding this?

BKPepe, 2019-06-19:
This one doesn't happen for me. I'm able to compile it locally for Turris Omnia (same arch) and it is even built on Turris servers - https://repo.turris.cz/master/omnia/packages/packages/python3-netifaces_0.10.9-3.7-1_arm_cortex-a9_vfpv3.ipk

Would be good if I can reproduce it.

cotequeiroz, 2019-06-19:
Can you compare your `build/temp.linux2-3.7/conftest.out` (from the package's build dir) with the one I posted above?

BKPepe, 2019-06-19:
Here it is:
```
✔ ~/work/turrisos40/omnia-hbd/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9/build/temp.linux2-3.7 [master L|…17] 
22:11 $ cat conftest.out
creating build/temp.linux2-3.7/build
creating build/temp.linux2-3.7/build/temp.linux2-3.7
ccache_cc -DNDEBUG -Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/pepe/work/turrisos40/omnia-hbd/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.2:Python-3.7.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/pepe/work/turrisos40/omnia-hbd/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/include -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include/fortify -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/python3.7 -fPIC -I/home/pepe/work/turrisos40/omnia-hbd/build/staging_dir/hostpkg/include/python3.7 -c build/temp.linux2-3.7/conftest-0.c -o build/temp.linux2-3.7/build/temp.linux2-3.7/conftest-0.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
ccache_cc build/temp.linux2-3.7/build/temp.linux2-3.7/conftest-0.o -L/usr/lib -o build/temp.linux2-3.7/conftest

```

cotequeiroz, 2019-06-19:
Thanks!  That helps, as it seems to be closer to gcc/binutils/toolchain than to python.  Nonetheless, it doesn't seem right that /usr/lib is used there.

commodo, 2019-06-20:
I'll take a look these days.

But
> Nonetheless, it doesn't seem right that /usr/lib is used there.

yes, it isn't right;
but this is too common for Python & Python packages;
the way all the build elements are stacked-up, some of these absolute/system paths [ /usr/lib, /usr/include, etc] sometimes slip through the cracks;
i guess that's one reason why the whole Python/Python3 packaging in OpenWrt looks so messy;

i sometimes like to re-post this article [ I did not write it ], because it resonates so much even today (7 years after it was written ) with Python packaging
http://lucumr.pocoo.org/2012/6/22/hate-hate-hate-everywhere/

jefferyto, 2019-06-20:
I tried compiling this, here is the relevant make output:

```
running build_ext
checking for getifaddrs...found.
checking for getnameinfo...found.
checking for IPv6 socket IOCTLs...not found.
checking for optional header files...netpacket/packet.h linux/atm.h linux/llc.h linux/tipc.h linux/dn.h.
checking whether struct sockaddr has a length field...no.
checking which sockaddr_xxx structs are defined...in in6 un ll atmpvc atmsvc dn llc.
checking for routing socket support...no.
checking for sysctl(CTL_NET...) support...no.
checking for netlink support...yes.
will use netlink to read routing table
building 'netifaces' extension
arm-openwrt-linux-muslgnueabi-gcc -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/media/jeff/Jekyll/Files/openwrt/testing/mvebu/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.2:Python-3.7.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/media/jeff/Jekyll/Files/openwrt/testing/mvebu/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include/fortify -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/python3.7 -fPIC -DNETIFACES_VERSION=0.10.9 -DHAVE_GETIFADDRS=1 -DHAVE_GETNAMEINFO=1 -DHAVE_NETPACKET_PACKET_H=1 -DHAVE_LINUX_ATM_H=1 -DHAVE_LINUX_LLC_H=1 -DHAVE_LINUX_TIPC_H=1 -DHAVE_LINUX_DN_H=1 -DHAVE_SOCKADDR_IN=1 -DHAVE_SOCKADDR_IN6=1 -DHAVE_SOCKADDR_UN=1 -DHAVE_SOCKADDR_LL=1 -DHAVE_SOCKADDR_ATMPVC=1 -DHAVE_SOCKADDR_ATMSVC=1 -DHAVE_SOCKADDR_DN=1 -DHAVE_SOCKADDR_LLC=1 -DHAVE_PF_NETLINK=1 -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/hostpkg/include/python3.7 -c netifaces.c -o build/temp.linux2-3.7/netifaces.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
creating build/lib.linux2-3.7
arm-openwrt-linux-muslgnueabi-gcc -shared -L/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/lib -L/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/lib -znow -zrelro -lpython3.7 -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/media/jeff/Jekyll/Files/openwrt/testing/mvebu/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include/fortify -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/python3.7 build/temp.linux2-3.7/netifaces.o -L/usr/lib -lpython3.7 -o build/lib.linux2-3.7/netifaces.cpython-37.so
```

And conftest.out:

```
creating build/temp.linux2-3.7/build
creating build/temp.linux2-3.7/build/temp.linux2-3.7
arm-openwrt-linux-muslgnueabi-gcc -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/media/jeff/Jekyll/Files/openwrt/testing/mvebu/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/Python-3.7.2:Python-3.7.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/media/jeff/Jekyll/Files/openwrt/testing/mvebu/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/netifaces-0.10.9:netifaces-0.10.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/mvebu/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/media/jeff/Jekyll/ [... cut ...] ttps://github.com/python/cpython/blob/v3.7.2/Lib/distutils/unixccompiler.py#L176-L179)), and whereas [`LDFLAGS` is added to `linker_so`](https://github.com/python/cpython/blob/v3.7.2/Lib/distutils/sysconfig.py#L194-L195), `linker_exec` only [gets the `CC` command](https://github.com/python/cpython/blob/v3.7.2/Lib/distutils/sysconfig.py#L217). As for why `LDFLAGS` is not added when linking executables... I have no idea :joy:

I also haven't been able to track down where/why `-L/usr/lib` is added.

cotequeiroz, 2019-06-21:
>I also haven't been able to track down where/why -L/usr/lib is added.

I have some answers, but I'm no expert in this.  I was just lucky and guessed them right :laughing:. 

I know the `/usr/lib` string was taken from `$(STAGING_DIR)/usr/lib/python3.7/_sysconfigdata.py:632`:
```
 'LIBDIR': '/usr/lib',
```
I'm guessing the file contents are generated somewhere.  Remember my note about being lucky.
If I change that to where my staging dir is, it works.
`LIBDIR` gets added to the `library_dirs` in `$(STAGING_DIR)/usr/lib/python3.7/distutils/command/build_ext.py:236`:
```
        # For building extensions with a shared Python library,
        # Python's library directory must be appended to library_dirs
        # See Issues: #1600860, #4366
        if (sysconfig.get_config_var('Py_ENABLE_SHARED')):
            if not sysconfig.python_build:
                # building third party extensions
                self.library_dirs.append(sysconfig.get_config_var('LIBDIR'))
            else:
                # building python standard extensions
                self.library_dirs.append('.')
```
I haven't guessed the answer to the `LDFLAGS` omission.  I don't like to push my luck that far.  Now the question becomes what's the best way to fix/patch/go around this?

cotequeiroz, 2019-06-21:
Found it.  Unlike Python2, Python3 does not apply `008-distutils-use-python-sysroot.patch`.

closed 2019-06-26 by cotequeiroz

jimmyd998, 2025-07-30:
Build noob here. I'm running a VM with latest version of Debian, along with Python3. The only error I get when updating my Openwrt Build environment is "WARNING: Makefile 'package/feeds/packages/mqttled/Makefile' has a dependency on 'python3-netifaces', which does not exist". Based on this thread, I'm trying to workout what I have to do to actually make this warning message go away. Any help would be appreciated.

BKPepe, 2025-07-30:
Your comment is not relevant to this PR anyhow. You should create a new PR. I am going to remove package `mqttled`, because the development is stalled and it is using archived repositories.
You can see it here https://github.com/trevortrevor/mqttled/issues/3 and as well with the refenced commit that netifaces package was removed.

PR #9288 (2019-06-26): python3: use _python_sysroot for cross-compilation
Maintainer: @commodo @jefferyto 
Compile tested: arm, WRT3200ACM, master - along with python3-netifaces
Run tested: none

Description:
This avoids the use of host header and libraries when building third-party extensions, as done in python2.

Fixes #9275