openwrt/packages #9283: various: package changes create dependency errors

EricLuehrsen, 2019-06-21:
Some recent commits have had some ugly config issue. It seems to be reoccurring with different packages.

cc: @tripolar @thess @micmac1 @stintel @neheb 
```
$ make menuconfig 
tmp/.config-package.in:60153:warning: ignoring unsupported character '@'
tmp/.config-package.in:60157:warning: ignoring unsupported character '@'
tmp/.config-package.in:60194:warning: ignoring unsupported character '@'
tmp/.config-package.in:60201:warning: ignoring unsupported character '@'
tmp/.config-package.in:60231:warning: ignoring unsupported character '@'
tmp/.config-package.in:102807:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:102807:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121556:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121556:	symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121571:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121571:	symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools
```
From temp config-package.in
```
			choice
				prompt "Selects Boost Runtime linkage."
				default boost-runtime-shared
				help
					Choose which C and C++ runtimes to use:
					-> Use Shared runtimes.
					-> Use Static runtimes.
						- Not available if Shared libs are to be built.
					-> Use both runtimes.
						- Not available if Shared libs are to be built.
						- Two separate versions of Boost are built, linking each to a different runtime.
						- This option requires "Use tagged names" option to be active.
	
				config boost-runtime-shared
					bool "Shared"
	
				config boost-runtime-static
					depends on @(!boost-shared-libs&&!boost-static-and-shared-libs)
					bool "Static"
	
				config boost-runtime-static-and-shared
					depends on @(boost-use-name-tags&&!boost-shared-libs&&!boost-static-and-shared-libs)
					bool "Both"
			endchoice
	
			choice
				prompt "Select a Variant."
				default boost-variant-release
				help
					Chooses which boost variant should be selected:
					-> Release: Optimizes Boost for release.
						- Optimization: Speed;  Debug Symbols: Off; Inlining: Full; Runtime Debugging: Off.
					-> Debug:
						- Optimization: Off; Debug Symbols: On; Inlining: Off; Runtime Debugging: On.
					-> Profile:
						- Profiling: On;  Debug Symbols: On.
	
				config boost-variant-release
					bool "Release"
	
				config boost-variant-debug
					bool "Debug"
	
				config boost-variant-profile
					bool "Profile"
			endchoice
	
			config boost-use-name-tags
				bool "Use tagged names."
				help
					Add name tags the lib files, to diferentiate each library version:
					  "-mt" for multi-threading.
					  "-d" for debugging.
					  "-s" for runtime static link".
					Might break compatibility with libraries that expect boost libs with default names.
				default n
	
			config boost-single-thread
				depends on @boost-use-name-tags
				bool "Single thread Support."
				help
					Compile Boost libraries in single-thread mode.
				default n
	
			config boost-build-type-complete
				depends on @boost-use-name-tags
				bool "Complete Boost Build."
				help
					Builds both release and debug libs. It will take much longer to compile.
				default n
		endmenu
```

micmac1, 2019-06-21:
Hi Eric,

Could you retry on latest trunk? jow and Nicolas pushed some config fixes yesterday.

Regards,
Seb

jow-, 2019-06-21:
> tmp/.config-package.in:60153:warning: ignoring unsupported character '@'

Update to latest master and issue a `make config-clean` to recompile kconfig.

jow-, 2019-06-21:
Actually the boost package is indeed broken and the recent kconfig update merely exposed it. The boost Makefile uses invalid raw kconfig syntax.

```diff
diff --git a/libs/boost/Makefile b/libs/boost/Makefile
index 9f0c0b473..7eed09583 100644
--- a/libs/boost/Makefile
+++ b/libs/boost/Makefile
@@ -196,11 +196,11 @@ define Package/boost/config
                                bool "Shared"
 
                        config boost-runtime-static
-                               depends on @(!boost-shared-libs&&!boost-static-and-shared-libs)
+                               depends on !boost-shared-libs&&!boost-static-and-shared-libs
                                bool "Static"
 
                        config boost-runtime-static-and-shared
-                               depends on @(boost-use-name-tags&&!boost-shared-libs&&!boost-static-and-shared-libs)
+                               depends on boost-use-name-tags&&!boost-shared-libs&&!boost-static-and-shared-libs
                                bool "Both"
                endchoice
 
@@ -237,14 +237,14 @@ define Package/boost/config
                        default n
 
                config boost-single-thread
-                       depends on @boost-use-name-tags
+                       depends on boost-use-name-tags
                        bool "Single thread Support."
                        help
                                Compile Boost libraries in single-thread mode.
                        default n
 
                config boost-build-type-complete
-                       depends on @boost-use-name-tags
+                       depends on boost-use-name-tags
                        bool "Complete Boost Build."
                        help
                                Builds both release and debug libs. It will take much longer to compile.
@@ -274,7 +274,7 @@ define Package/boost/config
                        select PACKAGE_boost-test
 
                config boost-coroutine2
-                       depends on !@GCC_VERSION_4_8
+                       depends on !GCC_VERSION_4_8
                        bool "Boost couroutine2 support."
                        select PACKAGE_boost-coroutine
                        default n
```

KanjiMonster, 2019-06-21:
Yepp, @jow-'s changes look correct. @ClaymorePT please have a look at it

ClaymorePT, 2019-06-21:
@EricLuehrsen those changes have nothing of recent. They have been added for over two years and they were checked at the time.

@jow- big thanks for noticing.

@KanjiMonster Looks good. Merge it!

EricLuehrsen, 2019-06-23:
`boost` package errors may have been hiding and waiting. The recursive dependencies are new, and this error seems to be a repeat occurrence this year with different packages.

PR #9293 (2019-06-23): boost: Makefile polishing and fix raw kconfig syntax
Maintainer: @ClaymorePT
Compile tested: Turris MOX, cortexa53, OpenWrt master

Description:
- Fix by @jow- in https://github.com/openwrt/packages/issues/9283#issuecomment-504320422. Authorship is kept. However, I need to Signed off it, otherwise DCO check will complain and probably CI, too.
- Makefile polishing

Fixes #9283
closed 2019-06-23 by neheb