openwrt/packages #9290: busybox: conflict between busybox and coreutils-sort

fantom-x, 2019-06-22:
Maintainer: @yousong
Environment: r7800 / master (using image builder)

Description:

```
 * check_data_file_clashes: Package coreutils-sort wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/sort
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-sort.
```

fantom-x, 2019-06-23:
These are all the coreutils-* packages that fail to install due to conflicts with busybox

```
Collected errors:
 * check_data_file_clashes: Package coreutils-basename wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/basename
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-basename.
 * check_data_file_clashes: Package coreutils-cut wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/cut
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-cut.
 * check_data_file_clashes: Package coreutils-dirname wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/dirname
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-dirname.
 * check_data_file_clashes: Package coreutils-du wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/du
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-du.
 * check_data_file_clashes: Package coreutils-env wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/env
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-env.
 * check_data_file_clashes: Package coreutils-expr wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/expr
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-expr.
 * check_data_file_clashes: Package coreutils-head wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/head
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-head.
 * check_data_file_clashes: Package coreutils-id wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/id
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-id.
 * check_data_file_clashes: Package coreutils-md5sum wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/md5sum
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-md5sum.
 * check_data_file_clashes: Package coreutils-mkfifo wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/mkfifo
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-mkfifo.
 * check_data_file_clashes: Package coreutils-printf wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/printf
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-printf.
 * check_data_file_clashes: Package coreutils-readlink wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/readlink
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-readlink.
 * check_data_file_clashes: Package coreutils-seq wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/seq
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-seq.
 * check_data_file_clashes: Package coreutils-sha256sum wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/sha256sum
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-sha256sum.
 * check_data_file_clashes: Package coreutils-sort wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/sort
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-sort.
 * check_data_file_clashes: Package coreutils-tail wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/tail
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-tail.
 * check_data_file_clashes: Package coreutils-tee wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/tee
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-tee.
 * check_data_file_clashes: Package coreutils-test wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/test
	But that file is already provided by package  * busybox
[... cut ...]
 * check_data_file_clashes: Package coreutils-yes wants to install file /build/openwrt-imagebuilder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/bin/yes
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package coreutils-yes.
Makefile:152: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:111: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:195: recipe for target 'image' failed
make: *** [image] Error 2

```

neheb, 2019-06-23:
ping @yousong

closed 2019-07-13 by yousong