openwrt/packages #9297: ruby-gems: unable to run gem command 'incompatible library version - stringio.so'

iain-taylor, 2019-06-23:
Maintainer: @luizluca 
Environment: (Lantiq xRX200 rev 1.2, BT Homehub 5A, OpenWrt 18.06.2 r7676-cddd7b4c77)
                     (also tested with OpenWrt SNAPSHOT r10192-3b36aae9d5)

Description: When trying to execute the gem command a LoadError (incompatible library version) for stringio.so is given and the command fails. This prevents use of gems on a mips platform. 

Comments in the Package OpenWrt Makefile (~lines 44-52) indicate this may be related to cross-compiling issues (x86-> mips) Issue has not been seen on an x86 test vm.

Research has shown a previous report elsewhere for an older build of openwrt agains a mips arduino platform indicates this may not be specifically related to the Lantiq xRX200 / BT Homehub 5A combination (See 'Install ruby-gems through opkg breaks ruby' https://forum.arduino.cc/index.php?topic=606406.0)

Testing thus far against the snapshot version of OpenWrt uses ruby 2.6.3p62 (2019-04-16 revision 67580) [mips-linux-gnu] indicates the issue persists beyond the 18.06.2 release.
Installation option is for the full ruby-stdlib package in both cases.

Error message as given when the gem command is executed
```
Traceback (most recent call last):
        6: from <internal:gem_prelude>:2:in `<internal:gem_prelude>'
        5: from <internal:gem_prelude>:2:in `require'
        4: from /usr/lib/ruby/2.6/rubygems.rb:116:in `<top (required)>'
        3: from /usr/lib/ruby/2.6/rubygems.rb:1365:in `<module:Gem>'
        2: from /usr/lib/ruby/2.6/rubygems.rb:1365:in `require'
        1: from /usr/lib/ruby/2.6/rubygems/specification.rb:18:in `<top (required)>'
/usr/lib/ruby/2.6/rubygems/specification.rb:18:in `require': incompatible library version - /usr/lib/ruby/2.6/mips-linux-gnu/stringio.so (LoadError)

```

luizluca, 2019-06-25:
Ruby checks for "incompatible modules" by looking if the undefined ruby_xmalloc symbol from the loaded module will be resolved as the one from libruby loaded by ruby. This should catch cases where the module was built for another ruby version (and uses another libruby file).

```
	/* Load file */
	if ((handle = (void*)dlopen(file, RTLD_LAZY|RTLD_GLOBAL)) == NULL) {
	    error = dln_strerror();
	    goto failed;
	}
# if defined RUBY_EXPORT
	{
	    void *ex = dlsym(handle, EXTERNAL_PREFIX"ruby_xmalloc");
	    if (ex && ex != ruby_xmalloc) {
```
https://github.com/ruby/ruby/blob/39ee412b14cbd60e108a2de62715c254f41f6bb2/dln.c#L1325-L1334

In my tests with ar71xx (mips_24kc), at the last line "if (ex &&..."  these two pointers are different. If I patch the code to comment that out, it simply works:

```
+            fprintf(stderr, "dln.c: %s\n", EXTERNAL_PREFIX"ruby_xmalloc");
             void *ex = dlsym(handle, EXTERNAL_PREFIX"ruby_xmalloc");
+            fprintf(stderr, "dln.c: ex=%p, ruby_xmalloc=%p\n", ex, ruby_xmalloc);
+            /*
              ...
```
And running it:
```
OpenWrt18.06# opkg install ruby ruby-stringio #libruby
OpenWrt18.06# echo 0 > /proc/sys/kernel/randomize_va_space
OpenWrt18.06# ruby -r stringio -e 'p StringIO.new("test123")'
dln.c: ruby_xmalloc
dln.c: ex=0x77c2a930, ruby_xmalloc=0x77dce36d
OpenWrt18.06#<StringIO:0x00432b10>
OpenWrt18.06# ldd /usr/lib/ruby/ruby2.5-bin 
	/lib/ld-musl-mips-sf.so.1 (0x77f5a000)
	libruby.so.2.5 => /usr/lib/libruby.so.2.5 (0x77d5c000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x77ce2000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77cbe000)
	libc.so => /lib/ld-musl-mips-sf.so.1 (0x77f5a000)
OpenWrt18.06# ldd /usr/lib/ruby/2.5/mips-linux-gnu/stringio.so
	ldd (0x77f5a000)
	libruby.so.2.5 => /usr/lib/libruby.so.2.5 (0x77d44000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x77cca000)
	libc.so => ldd (0x77f5a000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77ca6000)

Buildx86$ staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-nm build_dir/target-mips_24kc_musl/ruby-2.5.3/libruby.so.2.5.3 | grep ruby_xmalloc
0007236c T ruby_xmalloc

Buildx86$ staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-nm build_dir/target-mips_24kc_musl/ruby-2.5.3/.ext/mips-linux-gnu/stringio.so | grep ruby_xmalloc
         U ruby_xmalloc
```

So, ruby_xmalloc is undefined (as expected) in stringio.so.
Adding libruby.so and ldd addresses (I'm not an expert here, just guessing the math), it looks like ruby_xmalloc is correct:
```
             ruby_xmalloc=0x77dce36d
0x77d5c000 + 0x0007236c = 0x77dce36c
```

First question: "where does the 0x77c2a930 come from?"

Now the good part. I also patched stringio.c to simply print ruby_xmalloc address when it is initializing (it will run after dlopen() and dlsym()). Now addresses match, even before the code in stringio.c that uses ruby_xmalloc runs.

```
OpenWrt18.06# ruby -r stringio -e 'p StringIO.new("test123")'
dln.c: ruby_xmalloc
dln.c: ex=0x77dce36d, ruby_xmalloc=0x77dce36d
stringio.c: ruby_xmalloc=0x77dce36d
```
I even tried to remove RTLD_LAZY but it did not fix the issue.

The second question: "why dlsym() returned the wrong address if that symbol was already loaded from libruby?"

This is something that might get into symbol stripping, libc, hardening or even gcc. It is above "my paygrade". I'll need some help.

luizluca, 2019-06-26:
If I preload stringio.so, I can see that the strange address came from inside library address space:

```
OpenWrt18.06# LD_PRELOAD=/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so ldd /usr/lib/ruby/ruby2.5-bin 
	/lib/ld-musl-mips-sf.so.1 (0x77f5a000)
	/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so => /usr/lib/ruby/2.5/mips-linux-gnu/stringio.so (0x77f42000)
	libruby.so.2.5 => /usr/lib/libruby.so.2.5 (0x77d44000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x77cca000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77ca6000)
	libc.so => /lib/ld-musl-mips-sf.so.1 (0x77f5a000)
OpenWrt18.06# LD_PRELOAD=/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so ruby -rstringio -e 'p StringIO.new("test")' 
dln.c: ruby_xmalloc
dln.c:  ex=0x77f46930, ruby_xmalloc=0x77db639d, RTLD_NEXT=0, RTLD_DEFAULT=0x77db639d
#<StringIO:0x00432af8>
OpenWrt18.06# printf '0x%X\n' $((0x77d48930 - 0x77d44000))
0x4930
```

If I preload another lib (liblzo2.so), I can "move" the strange address:

```
OpenWrt18.06# LD_PRELOAD=/usr/lib/liblzo2.so:/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so ldd /usr/lib/ruby/ruby2.5-bin 
	/lib/ld-musl-mips-sf.so.1 (0x77f5a000)
	/usr/lib/liblzo2.so => /usr/lib/liblzo2.so (0x77f36000)
	/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so => /usr/lib/ruby/2.5/mips-linux-gnu/stringio.so (0x77f1e000)
	libruby.so.2.5 => /usr/lib/libruby.so.2.5 (0x77d20000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x77ca6000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77c82000)
	libc.so => /lib/ld-musl-mips-sf.so.1 (0x77f5a000)
OpenWrt18.06# LD_PRELOAD=/usr/lib/liblzo2.so:/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so ruby -rstringio -e 'p StringIO.new("test")' 
dln.c: ruby_xmalloc
dln.c:  ex=0x77f22930, ruby_xmalloc=0x77d9239d, RTLD_NEXT=0, RTLD_DEFAULT=0x77d9239d
#<StringIO:0x00432af8>
OpenWrt18.06# printf '0x%X\n' $((0x77f22930-0x77f1e000))
0x4930
```

I got the same offset 0x4930 from stringio.so address. And finally:

Buildx86$ staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-readelf -s staging_dir/target-mips_24kc_musl/root-ar71xx/usr/lib/ruby/2.5/mips-linux-gnu/stringio.so  | grep mall
    91: 00004930     0 FUNC    GLOBAL DEFAULT  UND ruby_xmalloc
   187: 00004930     0 FUNC    GLOBAL DEFAULT  UND ruby_xmalloc

So, dlsym() is returning the undefined entry in symbol table instead of trying to resolve the undefined symbol first and returning its address (as ruby code expects).

Who is to blame? ruby assumptions or musl dlsym()? http://git.musl-libc.org/cgit/musl/tree/ldso/dynlink.c#n2119?

luizluca, 2019-08-10:
I have some update. This patch fixes the issue: https://github.com/openwrt/openwrt/commit/b9674d528513c7c93205fa000fed7c0d3c6bb2e7

It is a nasty problem. I'll ask musl for validation before submitting it to OpenWrt core.

luizluca, 2019-08-16:
You can use this patch: luizluca/openwrt@a5b774c37feced8708d7f7b628fe7e6c5db40bf7:

I also built it for mips and mipsel:

- http://luizluca.github.io/openwrt/packages-18.06/mips_24kc/libc_1.1.19-1_mips_24kc.ipk
- http://luizluca.github.io/openwrt/packages-18.06/mipsel_24kc/libc_1.1.19-1_mipsel_24kc.ipk

But libc is strick to be updated and it breaks the system during the process. You can, however, manually asks for a libc:

mips:

```
# /usr/lib/ruby/ruby2.5-bin --disable-gems -r stringio -e 'puts 123'
Traceback (most recent call last):
-e:1:in `require': incompatible library version - /usr/lib/ruby/2.5/mips-linux-gnu/stringio.so (LoadError)

```
Problem is here:

```
# wget -q -O /tmp/libc.so http://luizluca.github.io/hacks/libc_mips.so && /tmp/libc.so /usr/lib/ruby/ruby2.5-bin --disable-gems -r stringio -e 'puts 123'
123
```

Problem is gone!

You just need to add '--disable-gems' if you did not install rubygems (that switch is added by sh script at /usr/bin/ruby)

Use http://luizluca.github.io/hacks/libc_mipsel.so if you are using little endian.

closed 2019-08-17 by jow-

luizluca, 2019-08-17:
The fix was commited. The next releases (18.06.x, 19.07.0, snapshot) will have it.

iain-taylor, 2019-11-02:
Howdo Luiz

Just a quick one – apologies for the time in getting back to you (Work has been crazy-busy!)

I’ve just had time to test on the actual hardware & everything looks good using the latest snapshot (02/11/2019)
Gems appear install etc exactly as they should… now I just need to get my ultimate use-case actually working, but that’s another story 😉

Thank you ever so much for your efforts – it’s been really appreciated (& skills way above my paygrade on the coding side)

All the best & thanks again

Iain

  *   http://luizluca.github.io/openwrt/packages-18.06/mips_24kc/libc_1.1.19-1_mips_24kc.ipk<https://apc01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fluizluca.github.io%2Fopenwrt%2Fpackages-18.06%2Fmips_24kc%2Flibc_1.1.19-1_mips_24kc.ipk&data=02%7C01%7C%7C1d29ded9f1154a4b4dab08d72201b317%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637015263002488427&sdata=4G5W%2Fd68kOHBqb9vuQvxXVYo0bOQj%2FrLMeW5mkcqCxk%3D&reserved=0>
  *   http://luizluca.github.io/openwrt/packages-18.06/mipsel_24kc/libc_1.1.19-1_mipsel_24kc.ipk<https://apc01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fluizluca.github.io%2Fopenwrt%2Fpackages-18.06%2Fmipsel_24kc%2Flibc_1.1.19-1_mipsel_24kc.ipk&data=02%7C01%7C%7C1d29ded9f1154a4b4dab08d72201b317%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637015263002498438&sdata=dEHZeHEK%2FMqpmoSsA%2F31tv5AEtAHvVxjgTxyEMFAQ%2FI%3D&reserved=0>