openwrt/packages #9300: various: recursive dependencies

EricLuehrsen, 2019-06-24:
We apologize for interrupting today's programming for these messages :smiley: but #9283 was only half resolved.
```
$ make menuconfig 
tmp/.config-package.in:103098:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:103098:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121863:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121863:	symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121878:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121878:	symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools
```
cc: @tripolar @thess @micmac1 @stintel @neheb @jow-

micmac1, 2019-06-24:
Maybe using "PROVIDES:=pulseaudio" _and_ "VARIANTs" in pulseaudio is too much? baresip seems the only package that uses the depend "pulseaudio". What if you change the depend in baresip from "pulseaudio" to "pulseaudio-daemon"? Does the error go away?

How about we change the depend in baresip to "pulseaudio-daemon", remove the "PROVIDES" line in "define Package/pulseaudio/Default" and add a "PROVIDES:=pulseaudio-daemon" to "define Package/pulseaudio-daemon-avahi"? Maybe that would stop confusing the build system.

micmac1, 2019-06-24:
Tried the above, doesn't work. What seems to work is this:

```
(--- a/sound/pulseaudio/Makefile
+++ b/sound/pulseaudio/Makefile
@@ -84,7 +84,7 @@ endef
 define Package/pulseaudio-tools
   SECTION:=sound
   CATEGORY:=Sound
-  DEPENDS:=+libsndfile +pulseaudio
+  DEPENDS:=+libsndfile pulseaudio
   TITLE:=Tools for Pulseaudio
   URL:=http://www.pulseaudio.org
   VARIANT:=noavahi
@@ -93,7 +93,7 @@ endef
 define Package/pulseaudio-profiles
   SECTION:=sound
   CATEGORY:=Sound
-  DEPENDS:=+pulseaudio
+  DEPENDS:=pulseaudio
   TITLE:=Profiles for Pulseaudio
   URL:=http://www.pulseaudio.org
 endef)
```

```
--- a/net/baresip/Makefile
+++ b/net/baresip/Makefile
@@ -199,7 +199,7 @@ $(eval $(call BuildPlugin,oss,OSS audio driver,oss,))
 $(eval $(call BuildPlugin,plc,Packet Loss Concealment,plc,+libspandsp))
 $(eval $(call BuildPlugin,portaudio,Portaudio audio driver,portaudio,+portaudio))
 $(eval $(call BuildPlugin,presence,Presence module,presence,))
-$(eval $(call BuildPlugin,pulse,Pulseaudio audio driver,pulse,+pulseaudio))
+$(eval $(call BuildPlugin,pulse,Pulseaudio audio driver,pulse,pulseaudio))
 $(eval $(call BuildPlugin,rtcpsummary,RTCP summary module,rtcpsummary,))
 $(eval $(call BuildPlugin,selfview,Video selfview module,selfview,))
 $(eval $(call BuildPlugin,sndfile,Audio dumper using libsndfile,sndfile,+libsndfile))
```

micmac1, 2019-06-24:
Actually for baresip I'd prefer this:

```
--- a/net/baresip/Makefile
+++ b/net/baresip/Makefile
@@ -199,7 +199,7 @@ $(eval $(call BuildPlugin,oss,OSS audio driver,oss,))
 $(eval $(call BuildPlugin,plc,Packet Loss Concealment,plc,+libspandsp))
 $(eval $(call BuildPlugin,portaudio,Portaudio audio driver,portaudio,+portaudio))
 $(eval $(call BuildPlugin,presence,Presence module,presence,))
-$(eval $(call BuildPlugin,pulse,Pulseaudio audio driver,pulse,+pulseaudio))
+$(eval $(call BuildPlugin,pulse,Pulseaudio audio driver,pulse,PACKAGE_$(PKG_NAME)-mod-pulse:pulseaudio))
 $(eval $(call BuildPlugin,rtcpsummary,RTCP summary module,rtcpsummary,))
 $(eval $(call BuildPlugin,selfview,Video selfview module,selfview,))
 $(eval $(call BuildPlugin,sndfile,Audio dumper using libsndfile,sndfile,+libsndfile))
```

This way pulseaudio doesn't get compiled automatically when building baresip with mod-pulse disabled. Anyway, just details.

Anybody has a better idea? Otherwise I'd send pull requests for pulseaudio and baresip.

linux-mate, 2019-06-25:
Another problem today
`tmp/.config-package.in:121862:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121862:  symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121877:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121877:  symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools
`

PR #9330 (2019-07-09): pulseaudio: fix recursive dependencies
Since OpenWrt commit e82a4d9cfb ("config: regenerate *_shipped sources")
a few recursive dependencies have been discovered that were not visible
before.

$ make menuconfig
tmp/.config-package.in:103098:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:103098:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121863:error: recursive dependency detected!
For a resolution refer to Docu [... cut]
commit ec3d0772 (2019-07-09): pulseaudio: fix recursive dependencies

Since OpenWrt commit e82a4d9cfb ("config: regenerate *_shipped sources")
a few recursive dependencies have been discovered that were not visible
before.

$ make menuconfig
tmp/.config-package.in:103098:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:103098:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121863:error: recursive dependency detected!
For a resolution refer to Doc [... cut]
commit 3d7ae8dd (2019-09-04): pulseaudio: fix recursive dependencies

Since OpenWrt commit e82a4d9cfb ("config: regenerate *_shipped sources")
a few recursive dependencies have been discovered that were not visible
before.

$ make menuconfig
tmp/.config-package.in:103098:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:103098:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121863:error: recursive dependency detected!
For a resolution refer to Doc [... cut]
closed 2019-07-09 by tripolar