openwrt/packages #9306: Openssh-sftp-server unknown.

Thiswarofmine, 2019-06-24:
Hello. Netgear wndr3800ch(mips_24kc). In 06/21/19 I could to install openssh-sftp-server easily. Now I can see an error(package unknown), even after opkg update. I looked in https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/packages/ and no openssh-sftp-server is there. It happened after 06/23 packages update (link abone). What to do? Where I can find it?

neheb, 2019-06-24:
ping @tripolar 

If we look at https://downloads.openwrt.org/releases/faillogs/mips_24kc/packages/openssh/without-pam/compile.txt , we see that it fails because of OpenSSL problens. ping @cotequeiroz 

My guess is that this commit broke things: https://github.com/openwrt/packages/commit/356bb11623efc17f17426f082bade51bc7c1a5d1

neheb, 2019-06-24:
It seems something is seriously wrong. https://github.com/openwrt/openwrt/blob/openwrt-18.06/package/libs/openssl/Config.in#L20 shows that WITH_DEPRECATED defaults to y. That commit removes it from being selected by haproxy. It should still be active though.

When looking at this for example: https://downloads.openwrt.org/releases/faillogs/mips_24kc/packages/alpine/ssl/compile.txt , it fails on a deprecated API...

cotequeiroz, 2019-06-24:
From https://downloads.openwrt.org/releases/faillogs/mips_24kc/packages/openssh/without-pam/compile.txt, it looks like openssh 7.7p1 is being compiled against openssl 1.1.1, which fails.  Note the reference to `pointer to incomplete type 'RSA {aka struct rsa_st}`.  It has been made opaque in 1.1.0.

cotequeiroz, 2019-06-24:
Looking for libopenssl in https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/base/ confirms it. It has built https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/base/libopenssl_1.1.1c-1_mips_24kc.ipk.

Thiswarofmine, 2019-06-24:
> Looking for libopenssl in https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/base/ confirms it. It has built https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/base/libopenssl_1.1.1c-1_mips_24kc.ipk.

Well. Maybe I was wrong with my explanations.
I connected to my router throught ssh. Its already has full openwrt firmware. Opkg update goes with no errors. When I type "opkg install openssh-sftp-server" I can see "package unknown". 2 days ago this operation was successful. Also, in mipc_24kc packages openssh-sftp-server was there. Now its clear

cotequeiroz, 2019-06-25:
You explanation was good; we were actually discussing the causes of the problem.

This should be fixed now (at least for mips_24kc):  https://downloads.openwrt.org/releases/packages-18.06/mipsel_24kc/packages/openssh-sftp-server_7.7p1-1_mipsel_24kc.ipk. Other platforms should get rebuilt and fixed as well within a day or so.  Tell me if you got it working, then you can close this issue.

Thiswarofmine, 2019-06-25:
Got it. Thanks for your help, sir!

Have a nice day.

Jimmy

cotequeiroz, 2019-06-26:
@jow- should get the credit, as he has done a great job at promptly fixing this major problem.  I'm closing this now.

closed 2019-06-26 by cotequeiroz