openwrt/packages #9307: youtube-dl does not use zip from tools/

neheb, 2019-06-24:
Maintainer: @BKPepe 
Environment: OpenWrt 18.06

https://downloads.openwrt.org/releases/faillogs/mips_24kc/packages/youtube-dl/compile.txt

It looks like this is trying to use zip from the host instead of zip from $(STAGING_DIR_HOST)/bin

BKPepe, 2019-07-08:
Even the youtube-dl is not updated to version 2019.7.2, the compile error is gone. :-/

```
root@OpenWrt:~# opkg install youtube-dl
Installing youtube-dl (2019.06.21-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.4/packages/mipsel_24kc/packages/youtube-dl_2019.06.21-1_mipsel_24kc.ipk
Configuring youtube-dl.
root@OpenWrt:~# youtube-dl
Usage: youtube-dl [OPTIONS] URL [URL...]

youtube-dl: error: You must provide at least one URL.
Type youtube-dl --help to see a list of all options.
```

However, I will update it just to be sure that this never gonna happen again.

neheb, 2019-07-08:
Yeah these errors are intermittent. Sometimes build failures related to missing zlib on the host show up.

I can't really test those as zlib is a dependency on too many things for me to remove.

closed 2019-08-04 by neheb