openwrt/packages #9308: ddns-scripts_route53-v1: failing with exit code 1 despite successful Route53 Updates.

supersebbo, 2019-06-24:
Maintainer: @chris5560, @maxberger
Environment: x86_64, 18.06.2
Running ddns-scripts 2.7.8-9 (snapshot version due to Curl SSL issue referenced below)
Running ddns-scripts 2.7.8-1 (tried running 2.7.8-9 from snapshot, however this seemed to break LUCI and was not displaying the username/password fields in the profile config)

Description:

After updating ddns_scripts to the current snapshot version to get over the [Curl SSL issues ](https://github.com/openwrt/packages/issues/6744) I am encountering further issues with the package.

When the update script is called, it is successfully updating Route53 but then the script is exiting on an error code and killing the ddns update process. 

I can re-create this at will by logging into the R53 console and changing the record in question to an arbitrary IP then waiting for the next DDNS check.

The DDNS script runs fine until an update is needed, then errors, despite a successful DDNS update. 

I've annotated the script log below:

```
##### This is the first time I ran the it after creating the ddns profile. 
 173951       : ************ ************** ************** **************
 173951  note : PID '26169' started at 2019-06-24 17:39
 173951       : ddns version  : 2.7.8-9
 173951       : uci configuration:
ddns.nvr.cacert='/etc/ssl/certs'
ddns.nvr.domain='REDACTED'
ddns.nvr.enabled='1'
ddns.nvr.lookup_host='my.domain.com'
ddns.nvr.password='***PW***'
ddns.nvr.service_name='route53-v1'
ddns.nvr.use_https='1'
ddns.nvr.username='REDACTED'
ddns.nvr=service
 173951       : verbose mode  : 0 - run normal, NO console output
 173951       : check interval: 600 seconds
 173951       : force interval: 259200 seconds
 173951       : retry interval: 60 seconds
 173951       : retry counter : 0 times
 173951       : No old process
 173951       : last update: never
 173951       : Detect registered/public IP
 173951       : #> /usr/bin/nslookup my.domain.com  >/var/run/ddns/nvr.dat 2>/var/run/ddns/nvr.err
 173952  WARN : NO valid IP found
 173952  info : Starting main loop at 2019-06-24 17:39
 173952       : Detect local IP on 'network'
 173952       : Local IP '77.99.xxx.yyy' detected on network 'wan'
##### Correctly reports no remote IP was configured (actually it was set to 0.0.0.0 but dnsmasq blocked this as a rebind attempt) so an update is called. 
 173952       : Update needed - L: '77.99.xxx.yyy' <> R: ''
 173952       : parsing script '/usr/lib/ddns/update_route53_v1.sh'
 173953       : <?xml version="1.0"?>
<ChangeResourceRecordSetsResponse xmlns="https://route53.amazonaws.com/doc/2013-04-01/"><ChangeInfo><Id>/change/REDACTED</Id><Status>PENDING</Status><SubmittedAt>2019-06-24T16:39:53.181Z</SubmittedAt></ChangeInfo></ChangeResourceRecordSetsResponse>
##### The recordset is successfully updated in R53 (verified through the R53 console)
 173953  info : Update successful - IP '77.99.229.198' send
 173953  info : Forced update successful - IP: '77.99.229.198' send
##### However the script exits with error 1
 173953  WARN : PID '26169' exit WITH ERROR '1' at 2019-06-24 17:39
##### So I manually restarted the DDNS process for this domain.
 174117       : ************ ************** ************** **************
 174117  note : PID '29364' started at 2019-06-24 17:41
 174117       : ddns version  : 2.7.8-9
 174117       : uci configuration:
ddns.nvr.cacert='/etc/ssl/certs'
ddns.nvr.domain='REDACTED'
ddns.nvr.enabled='1'
ddns.nvr.lookup_host='my.domain.com'
ddns.nvr.password='***PW***'
ddns.nvr.service_name='route53-v1'
ddns.nvr.use_https='1'
ddns.nvr.username='REDACTED'
ddns.nvr=service
 174117       : verbose mode  : 0 - run normal, NO console output
 174117       : check interval: 600 seconds
 174117       : force interval: 259200 seconds
 174117       : retry interval: 60 seconds
 174117       : retry counter : 0 times
 174117       : No old process
 174117       : last update: 2019-06-24 17:39
 174117       : Detect registered/public IP
 174117       : #> /usr/bin/nslookup my.domain.com  >/var/run/ddns/nvr.dat 2>/var/run/ddns/nvr.err
 174117       : Registered IP '77.99.xxx.yyy' detected
 174117  info : Starting main loop at 2019-06-24 17:41
 174117       : Detect local IP on 'network'
 174117       : Local IP '77.99.xxx.yyy' detected on network 'wan'
 174117       : Waiting 600 seconds (Check Interval)
##### Everything is fine for a few iterations here because, despite erroring, the script successfully updated Route53 at 17:39:53
 175117       : Detect registered/public IP
 175117       : #> /usr/bin/nslookup my.domain.com >/var/run/ddns/nvr.dat 2>/var/run/ddns/nvr.err
 175117       : Registered IP '77.99.xxx.yyy' detected
 175117  info : Rerun IP check at 2019-06-24 17:51
 175117       : Detect local IP on 'network'
 175117       : Local IP '77.99.xxx.yyy' detected on network 'wan'
 175117       : Waiting 600 seconds (Check Interval)
 180117       : Detect registered/public IP
 180117       : #> /usr/bin/nslookup nvr.sdjb.net  >/var/run/ddns/nvr.dat 2>/var/run/ddns/nvr.err
 180117       : Registered IP '77.99.xxx.yyy' detected
 180117  info : Rerun IP check at 2019-06-24 18:01
 180117       : Detect local IP on 'network'
 180118       : Local IP '77.99.xxx.yyy' detected on network 'wan'
 180118       : Waiting 600 seconds (Check Interval)
##### At this point I logged into the AWS console and changed the A record to a random IP to force an update. 
 181118       : Detect registered/public IP
 181118       : #> /usr/bin/nslookup my.domain.com  >/var/run/ddns/nvr.dat 2>/var/run/ddns/nvr.err
 181118       : Registered IP '77.99.xxx.zzz' detected
##### Unclear what this WARN is referencing
 181118  WARN : Updating IP at DDNS provider failed - starting retry 1/0
 181118       : Detect local IP on 'network'
 181118       : Local IP '77.99.xxx.yyy' detected on network 'wan'
 181118       : Update needed - L: '77.99.xxx.yyy' <> R: '77.99.xxx.zzz'
 181118       : parsing script '/usr/lib/ddns/update_route53_v1.sh'
 181119       : <?xml version="1.0"?>
<ChangeResourceRecordSetsResponse xmlns="https://route53.amazonaws.com/doc/2013-04-01/"><ChangeInfo><Id>/change/REDACTED</Id><Status>PENDING</Status><SubmittedAt>2019-06-24T17:11:19.171Z</SubmittedAt></ChangeInfo></ChangeResourceRecordSetsResponse>
 181119  info : Update successful - IP '77.99.xxx.yyy' send
 181119  info : Forced update successful - IP: '77.99.xxx.yyy' send
##### Once again, despite successful update, the script exits with error and needs to be manually restarted. 
 181119  WARN : PID '29364' exit WITH ERROR '1' at 2019-06-24 18:11
```

dibdot, 2019-06-26:
Please don't mix different versions and open tickets based on that. Please always test against latest master/snapshot ... if you have an LuCI issue open a ticket at there.

closed 2019-06-26 by dibdot

supersebbo, 2019-06-28:
Ok.. so perhaps I don't understand how this should be addressed. 

There is currently no way for regular users (running stable builds) to use this package.  It's broken in the release.  It's broken in the snapshot packages.  Do we just have to wait for the next release and hope it will be fixed without any issue to track it?