openwrt/packages #9310: opkg upgrade *: upgrade leaves router unbootable

miri64, 2019-06-24:
On a freshly flashed [WNDR3700v2](https://openwrt.org/toh/netgear/wndr3700) (OpenWrt 18.06.2) I was not able to reboot after I upgraded all packages in `opkg list-upgradable`. Sadly, after a day of debugging I wasn't able to pin-point the packet causing the issue. Here are some observation I had during upgrading:

# 1. getcwd() failed no such file or directory

When doing

```
opkg update
opkg upgrade opkg
opkg upgrade libuclient uclient-fetch
opkg upgrade uci libuci
opkg update
```

directly after the SSH server becomes accessible, I sometimes get `getcwd() failed no such file or directory`. After I login again it is gone

# 2. /etc/rc.common: eval: line 117: uci_load_validate: not found

When upgrading `busybox` I get the output

```
/etc/rc.common: eval: line 117: uci_load_validate: not found
```

I also get this output for some other packages (sorry I'm done re-bricking my router to look up which packets exactly) in a bulk update.

# 3. some packets are not downloadable, upgrade stops
In the middle of a `opkg upgrade ...` `opkg` says the http server is not reachable. This can be solved by installing the packages that are listed first.

# 4. Upgrading `libjson-c` causes several services to not run anymore
`luci` and `dnsmasq` are failing to start when `libjson-c` is updated (`libjson-c.so.2` is deleted and replaced by `libjson-c.so.4`, which is linked by `jsonfilter` and `ubus`). Upgrading the library first and then

```
opkg --force-reinstall install jsonfilter ubus
```

# 5. netifd comes not up again
```
opkg upgrade netifd
```

causes the daemon not to come up again. Sometimes even in a tmux session and not even after reboot if it is the first thing upgraded after a freshly reset image.

# 6. Script helps, but does not solve the problem
https://github.com/tavinus/opkg-upgrade helps with reducing errors (when addressing 3 and 4 first it passes without any errors), but the system still is not bootable.

Sorry for not being too specific, but after an unfruitful day of trying to find out what went wrong I just decided not to upgrade after all and roll with just the bare 18.06.2.

devosb, 2019-06-25:
I think I have seen the same problem, at least for items 4 and 5 above. If not, my apologies for not understanding the situation. I have a TP-Link ArcherC7 v5.0, Yesterday (24 Jun 2019 at 10:30 UTC) it was updated (as it is everyday at that time), and then would not route packets or respond to SSH. A LAN ethernet port was still active, I could see that status using another OpenWRT router, but could not access the system. I used TFTP to reinstall OpenWRT 18.06.1, then upgraded to 18.06.2, and attempted to get the many updates that were found.

The upgrade for netifd (from 2018-11-19-4b83102d-2 to 2019-06-15-9932ed02-1) caused the router to be unresponsive. The output from `opkg upgrade netifd` was
```
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/netifd_2019-06-15-9932ed02-1_mips_24kc.ipk
'radio0' is disabled
'radio1' is disabled
```
I thought maybe there was an issue with having the WiFi disabled (I use a wired connection) but repeating the process (after using TFTP again) produced the same result when the WiFi was enabled.

Using TFTP yet again I tried to get updates minus `netifd`. Upgrading `logd` (from 2018-02-14-128bc35f-2 to 2019-06-16-4df34a4d-2) resulted in errors concerning `libjson` as you can see here
```
Upgrading logd on root from 2018-02-14-128bc35f-2 to 2019-06-16-4df34a4d-2...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/logd_2019-06-16-4df34a4d-2_mips_24kc.ipk
Error loading shared library libjson-c.so.2: No such file or directory (needed by /usr/bin/jshn)
Error loading shared library libjson-c.so.2: No such file or directory (needed by /lib/libblobmsg_json.so)
Error relocating /lib/libblobmsg_json.so: array_list_length: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_array: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_put: symbol not found
Error relocating /lib/libblobmsg_json.so: array_list_get_idx: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_string: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_type: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_double: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_boolean: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_int: symbol not found
Error relocating /lib/libblobmsg_json.so: json_tokener_parse: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_from_file: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_object: symbol not found
Error relocating /usr/bin/jshn: json_object_get_object: symbol not found
Error relocating /usr/bin/jshn: json_object_new_int64: symbol not found
Error relocating /usr/bin/jshn: json_object_to_json_string: symbol not found
Error relocating /usr/bin/jshn: json_object_get_int64: symbol not found
Error relocating /usr/bin/jshn: json_tokener_parse: symbol not found
Error relocating /usr/bin/jshn: json_object_new_boolean: symbol not found
Error relocating /usr/bin/jshn: json_object_new_double: symbol not found
Error relocating /usr/bin/jshn: json_object_new_array: symbol not found
Error relocating /usr/bin/jshn: json_object_get_boolean: symbol not found
Error relocating /usr/bin/jshn: json_object_get_double: symbol not found
Error relocating /usr/bin/jshn: json_object_new_string: symbol not found
Error relocating /usr/bin/jshn: json_object_get_type: symbol not found
Error relocating /usr/bin/jshn: json_object_get_string: symbol not found
Error relocating /usr/bin/jshn: json_object_object_add: symbol not found
Error relocating /usr/bin/jshn: json_object_array_add: symbol not found
Error relocating /usr/bin/jshn: array_list_get_idx: symbol not found
Error relocating /usr/bin/jshn: json_object_put: symbol not found
Error relocating /usr/bin/jshn: json_object_get_array: symbol not found
Error relocating /usr/bin/jshn: array_list_length: symbol not found
Error relocating /usr/bin/jshn: json_object_new_object: symbol not found
Error loading shared library libjson-c.so.2: No such file or directory (needed by /bin/ubus)
Error loading shared library libjson-c.so.2: No such file or directory (needed by /lib/libblobmsg_json.so)
Error relocating /lib/libblobmsg_json.so: array_list_length: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_array: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_put: symbol not found
Error relocating /lib/libblobmsg_json.so: array_list_get_idx: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_string: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_type: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_double: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_boolean: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_int: symbol not found
Error relocating /lib/libblobmsg_json.so: json_tokener_parse: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_from_file: symbol not found
Error relocating /lib/libblobmsg_json.so: json_object_get_object: symbol not found
```
The updates completed (except for `netifd`) but after rebooting the router is was unresponsive. This was before upgrading `libjson-script` and `libblobmsg-json`.

miri64, 2019-06-25:
> I think I have seen the same problem, at least for items 4 and 5 above. If not, my apologies for not understanding the situation.

Yepp, looks pretty much like the problems I had.

jow-, 2019-06-25:
A series of faults in the CI system led to broken upgrade repositories (packages built from the wrong branch and being binary-incompatible to 18.06.x). Since this is not related to the packages repo but to the CI infrastructure, I'll close this ticket here.

Please follow up on https://bugs.openwrt.org/index.php?do=details&task_id=2339 , http://lists.infradead.org/pipermail/openwrt-devel/2019-June/017808.html and https://forum.openwrt.org/t/openwrt-org-server-configuration-403-forbidden/39486/12

closed 2019-06-25 by jow-

miri64, 2019-06-27:
Thanks for the information and the good work put into in fixing the problem @jow! Updates are working now again again as expected _and_ the router is able to reboot (never was so scared in my life to hit enter though :wink:)