openwrt/packages #9314: libgpg-error compile error with gawk-5.0+

nobk, 2019-06-24:
Device problem occurs on build host with gawk 5.0+ installed.
openwrt package git trunk
libgpg-error 

Steps to reproduce
1. upgrade host gawk to 5.0+
2. make package/libgpg-error/clean
3. make -j1 V=s package/libgpg-error/compile
```
--CUT--
./mkerrcodes | gawk -f ./mkerrcodes2.awk >code-from-errno.h
gawk: ./mkerrcodes2.awk:94: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
        ./err-sources.h.in >err-sources-sym.h
gawk: ./mkstrtable.awk:113: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
        ./err-codes.h.in >err-codes-sym.h
gawk: ./mkstrtable.awk:113: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
        -v prefix=GPG_ERR_ -v namespace=errnos_ \
        ./errnos.in >errnos-sym.h
gawk: fatal: cannot use gawk builtin `namespace' as variable name
make[5]: *** [Makefile:1616: errnos-sym.h] Error 2
--CUT--
```

nobk, 2019-06-25:
find bug report https://dev.gnupg.org/T4459 also.
and Gentoo including patches https://bugs.gentoo.org/683254

neheb, 2019-07-02:
Resubmitted: https://github.com/openwrt/packages/pull/9371

closed 2019-07-11 by nobk