openwrt/packages #9318: buildpr circleci hook is failing

James-TR, 2019-06-25:
The buildpr circleci hook is failing with the below error:

#!/bin/bash -eo pipefail
curl "https://$SDK_HOST/$SDK_PATH/sha256sums" -sS -o sha256sums
curl "https://$SDK_HOST/$SDK_PATH/sha256sums.asc" -sS -o sha256sums.asc
gpg --with-fingerprint --verify sha256sums.asc sha256sums
rsync -av "$SDK_HOST::downloads/$SDK_PATH/$SDK_FILE" .
sha256sum -c --ignore-missing sha256sums
gpg: Signature made Sat Jun 22 17:58:09 2019 UTC
gpg:                using RSA key F93525A88B699029
gpg: checking the trustdb
gpg: marginals needed: 3  completes needed: 1  trust model: pgp
gpg: depth: 0  valid:   3  signed:   0  trust: 0-, 0q, 0n, 0m, 0f, 3u
gpg: next trustdb check due at 2020-05-15
gpg: BAD signature from "LEDE Build System (LEDE GnuPG key for unattended build jobs) <lede-adm@lists.infradead.org>" [ultimate]
Exited with code 1

neheb, 2019-06-25:
Known issue: http://lists.infradead.org/pipermail/openwrt-devel/2019-June/017808.html

jow-, 2019-06-26:
I'm not actually sure that the CircleCI issue is related to the other problems we had with the repo. Right now it appears to choke on signature checking but when I was verifying things manually here, the PGP signatures were fine...

James-TR, 2019-06-26:
Some pull requests made after opening this issue have succeeded. I'll see when I get home if it's fixed

closed 2019-07-25 by James-TR