openwrt/packages #9326: seafile-seahub: Broken Dependencies - flup

jeffsf, 2019-06-26:
2019-06-26

* git pull [master]
* scripts/feeds update -a
* scripts/feeds install -a

diffconfig, after pusleaudio errors, returns
```
CONFIG_TARGET_ar71xx=y
CONFIG_TARGET_ar71xx_generic=y
CONFIG_TARGET_ar71xx_generic_DEVICE_mr16=y
```
make indicates 
```
WARNING: Makefile 'package/feeds/packages/seafile-seahub/Makefile' has a dependency on 'flup', which does not exist
 make[1] clean
WARNING: Makefile 'package/feeds/packages/seafile-seahub/Makefile' has a dependency on 'flup', which does not exist
make[2]: Entering directory '/home/jeff/devel/openwrt/scripts/config'
make[2]: Leaving directory '/home/jeff/devel/openwrt/scripts/config'
```

cotequeiroz, 2019-06-26:
I've tried to reproduce your problem, to no avail.  I've got your diffconfig, with the pulseaudio errors, but `make clean` turns out clean, and `make` starts to do its job--while I did not finish the full build, the warning should come up before it actually starts.  Try to run `rm -rf tmp; make config-clean` and try again.

jeffsf, 2019-06-26:
Reported by OP at https://forum.openwrt.org/t/meraki-mr16-noob-to-unix-please-help/39172/22?u=jeff and locally confirmed with my existing build tree.

```
jeff@deb-devel:~/devel/openwrt$ make menuconfig
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/feeds/packages/seafile-seahub/Makefile' has a dependency on 'flup', which does not exist
tmp/.config-package.in:102870:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:102870:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121610:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121610:	symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121625:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121625:	symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

jeff@deb-devel:~/devel/openwrt$ ./scripts/diffconfig.sh 
tmp/.config-package.in:102870:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:102870:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121610:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121610:	symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121625:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121625:	symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools
tmp/.config-package.in:102870:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:102870:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121610:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121610:	symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121625:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121625:	symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools
CONFIG_TARGET_ar71xx=y
CONFIG_TARGET_ar71xx_generic=y
CONFIG_TARGET_ar71xx_generic_DEVICE_mr16=y
jeff@deb-devel:~/devel/openwrt$ make -j12 clean download world
WARNING: Makefile 'package/feeds/packages/seafile-seahub/Makefile' has a dependency on 'flup', which does not exist
 make[1] clean
^C/home/jeff/devel/openwrt/include/toplevel.mk:193: recipe for target 'clean' failed
make: *** [clean] Interrupt
```

jeffsf, 2019-06-26:
```
jeff@deb-devel:~/devel$ git clone https://git.openwrt.org/openwrt/openwrt.git openwrt-tmp
jeff@deb-devel:~/devel$ cd openwrt-tmp/
jeff@deb-devel:~/devel/openwrt-tmp$ ./scripts/feeds update -a
jeff@deb-devel:~/devel/openwrt-tmp$ cat > .config
CONFIG_TARGET_ar71xx=y
CONFIG_TARGET_ar71xx_generic=y
CONFIG_TARGET_ar71xx_generic_DEVICE_mr16=y
jeff@deb-devel:~/devel/openwrt-tmp$ make menuconfig
Collecting package info: done
WARNING: Makefile 'package/feeds/packages/seafile-seahub/Makefile' has a dependency on 'flup', which does not exist
tmp/.config-package.in:103149:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:103149:	symbol PACKAGE_baresip-mod-pulse depends on PACKAGE_baresip-mod-pulse
tmp/.config-package.in:121935:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121935:	symbol PACKAGE_pulseaudio-profiles depends on PACKAGE_pulseaudio-profiles
tmp/.config-package.in:121950:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:121950:	symbol PACKAGE_pulseaudio-tools depends on PACKAGE_pulseaudio-tools
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

jeff@deb-devel:~/devel/openwrt-tmp$ 
```

cotequeiroz, 2019-06-26:
Got it.  flup was renamed python-flup; seafile-seahub needs to adjust.
```
diff --git a/net/seafile-seahub/Makefile b/net/seafile-seahub/Makefile
index 50a4dff..edc05f3 100644
--- a/net/seafile-seahub/Makefile
+++ b/net/seafile-seahub/Makefile
@@ -35,7 +35,7 @@ define Package/seafile-seahub
        +django-compressor +django-formtools +django-jsonfield \
        +django-picklefield +django-postoffice +django-restframework \
        +django-simple-captcha +django-statici18n +django-webpack-loader \
-       +flup +gunicorn +openpyxl \
+       +python-flup +gunicorn +openpyxl \
        $(foreach dep,$(SEAFILE_PYTHON_DEPENDS),+python-$(dep))
 endef
 
```

PR #9328 (2019-06-27): seafile-seahub: adjust to flup rename
Maintainer: @commodo 
Compile tested: `make menu` tested (this is a trivial change)
Run tested: none

Description:
This adjusts seafile-seahub DEPENDS to flup package rename to python-flup.  Increasing `PKG_RELEASE` since this changes the package's install-time dependencies.

Fixes #9326
closed 2019-06-27 by BKPepe