openwrt/packages #9342: transmission: all torrents lost after system reboot

swiftech, 2019-06-27:
transmission-daemon-openssl	2.94-1
OpenWrt 18.06.2 r7676-cddd7b4c77

1. open transmission-web in browser.
2. upload one or more magnet url.
3. it begins transferring.
4. reboot system by: 
    $ reboot
5. after rebooting, open transmission-web again, no torrents left, it's empty there. check /tmp/transmission/torrents dir, no files exist.

PS: rebooting transmission-daemon is fine, only rebooting system will make this happen.

neheb, 2019-06-27:
During reboot, OpenWrt kills transmission as it takes too long to terminate. Not sure if there's a way to make OpenWrt wait for transmission.

swiftech, 2019-06-27:
but it won't happen on Lubuntu, in my another pi device, even cut off it's power directly.

neheb, 2019-06-27:
Maybe the data is written to disk faster.

lucize, 2019-06-27:
> /tmp/transmission/torrents

is not persistent, is a tmpfs

swiftech, 2019-06-29:
@lucize yes, what you said is right. Changing the config_dir from /tmp/transmission to another permanent dir solved this problem, thanks for reminding me about that.

closed 2019-06-30 by swiftech