openwrt/packages #9346: shadowsocks-libev-server: weird port change

mkyybx, 2019-06-28:
Maintainer: @yousong
Environment:
arch: armv5tel Feroceon 88FR131 rev 1 (v5l)
model: Linksys EA3500
OpenErt: 18.06.2

Description:

Please see [issue 2412](https://github.com/shadowsocks/shadowsocks-libev/issues/2412) of shadowsocks/shadowsocks-libev. Thanks.

yousong, 2019-06-28:
I am quoting from 6.3.2.3.7 of iso c lang standard draft.

```
A pointer to an object type may be converted to a pointer to a different object type. If the
resulting pointer is not correctly aligned) for the referenced type, the behavior is
undefined...
```

I think we should fix the code in question.  That is, src/server.c:897

```
port = (*(uint16_t *)(server->buf->data + offset));
```

Do you feel like providing a patch?

yousong, 2019-06-28:
Link to the quoted document: http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1570.pdf

yousong, 2019-06-28:
I just noticed that similar issues also exist in other parts of the shadowsocks-libev codebase.

```
➜  ~/git-repo/shadowsocks-libev git:(master) ag  '\*\([^)]+ \*\)'
libbloom/murmur2/MurmurHash2.c
33:             unsigned int k = *(unsigned int *)data;

src/aead.c
594:    mlen = ntohs(*(uint16_t *)len_buf);

src/server.c
806:                addr->sin_addr = *(struct in_addr *)(server->buf->data + offset);
815:            addr->sin_port   = *(uint16_t *)(server->buf->data + offset);
823:            uint8_t name_len = *(uint8_t *)(server->buf->data + offset);
845:                    addr->sin_port   = *(uint16_t *)(server->buf->data + offset);
853:                    addr->sin6_port   = *(uint16_t *)(server->buf->data + offset);
873:                addr->sin6_addr = *(struct in6_addr *)(server->buf->data + offset);
883:            addr->sin6_port  = *(uint16_t *)(server->buf->data + offset);
897:        port = (*(uint16_t *)(server->buf->data + offset));

src/uthash.h
583:#define get16bits(d) (*((const uint16_t *) (d)))

src/local.c
384:            uint16_t p = ntohs(*(uint16_t *)(buf->data + request_len + in_addr_len));
393:        uint8_t name_len = *(uint8_t *)(buf->data + request_len);
403:                ntohs(*(uint16_t *)(buf->data + request_len + 1 + name_len));
417:            uint16_t p = ntohs(*(uint16_t *)(buf->data + request_len + in6_addr_len));
440:    uint16_t dst_port = ntohs(*(uint16_t *)(abuf->data + abuf->len - 2));

src/udprelay.c
244:    const uint8_t atyp = *(uint8_t *)buf;
255:                addr->sin_addr   = *(struct in_addr *)(buf + offset);
256:                addr->sin_port   = *(uint16_t *)(buf + offset + in_addr_len);
266:        uint8_t name_len = *(uint8_t *)(buf + offset);
276:                        addr->sin_port   = *(uint16_t *)(buf + offset + 1 + name_len);
281:                        addr->sin6_port   = *(uint16_t *)(buf + offset + 1 + name_len);
298:                addr->sin6_addr   = *(struct in6_addr *)(buf + offset);
299:                addr->sin6_port   = *(uint16_t *)(buf + offset + in6_addr_len);
315:        sprintf(port, "%d", ntohs(*(uint16_t *)(buf + offset)));
969:    uint8_t frag = *(uint8_t *)(buf->data + 2);
```

uint8 should be fine

mkyybx, 2019-06-28:
I will look into this this weekend and my possible solution is adding padding to the variable length fields to make sure they are at least 16-bit(half word) aligned.

yousong, 2019-06-28:
I am not sure, but it seems to me that `buf->data` are user-controlled input.  Maybe it needs something like `get_unaligned()` of the linux kernel.

mkyybx, 2019-06-28:
Please refer to pull request [#2414](https://github.com/shadowsocks/shadowsocks-libev/pull/2414) of shadowsocks-libev.

yousong, 2019-06-29:
@mkyybx Thank you for taking care of this!

From the latest response there, I think we agreed that the issue needs to be fixed on the c code, not the compiler.

closed 2019-06-30 by mkyybx

yousong, 2019-08-12:
Hi @mkyybx ,

I am writing to let you know that shadowsocks-libev in master branch has just been updated to version 3.3.1 (5d7d44d605dc71c3edf157bae3fe35f07a3f516f).  This version should fix the issue here.

Thank you for the efforts.