openwrt/packages #9348: opus 1.3.1 bad performance on MIPS

dddaniel, 2019-06-28:
I noticed a pretty bad performance decrease in 'opus_multistream_encode'
when using v1.3.1 on a MIPS platform.
This issue is only present with v1.3.1.
v1.3 and previous versions work fine.

I am in the process of collecting details and reporting the issue to the opus team.

Would you please consider downgrading back to 1.3 until this issue is fixed ?

neheb, 2019-06-28:
I prefer to patch 1.3.1 instead. Have you bisected the issue?

dddaniel, 2019-06-28:
> I prefer to patch 1.3.1 instead. Have you bisected the issue?

Not yet, but that's on my todo list for tomorrow.
An Issue has already been opened in the opus project: https://github.com/xiph/opus/issues/139

diizzyy, 2019-08-01:
Have you tried disabling MIPS16 and/or adding -O2?

neheb, 2019-08-02:
Disabling MIPS16 should be done on opus IMO.

diizzyy, 2019-08-22:
Closing this due to timeout

closed 2019-08-22 by diizzyy